eblot/pyftdi #27: No examples available for FTDI module.

bicepjai, 2015-04-14:
I tried to use this module for working with 
http://www.sainsmart.com/sainsmart-4-channel-5v-usb-relay-board-module-controller-for-automation-robotics.html
but could now find any examples.

cospan, 2015-04-14:
That's a pretty cool board!

I've been using this library for a while now, specifically to toggle GPIOs, program SPI flash chips and send data to and from my FPGA using the synchronous FIFOs. I would like to help you out but unfortunately I don't have the time to read through all the documentation for that board.

If any of your needs are related to anything above I can help out with an example or two.

eblot, 2015-09-28:
I'm not sure what is a "generic FTDI device" here?
Could you provide the kind of FTDI device you use, and how they are reported on the USB bus? (vendor ID, product ID, ...)

karelv, 2016-03-19:
@cospan, As you stated, that you use this lib for GPIO, may you be so kind to share a basic example about how to:
1. Configure the CBUS0&1 to be output and CBUS2&3 to be input (as GPIO)
2. demo program to write CBUS0 to 'zero', CBUS1 to 'one'.
3. demo program to read CBUS2&3.

This would be very helpful for me!

Thanks in advance,
Karel.

cospan, 2016-03-23:
Here's a simple python module and command line tool I wrote to control GPIOs for a FT2232H on one of my boards.

https://github.com/CospanDesign/python/tree/master/ftdi/gpio

There are two files:
gpio_controller.py: Python module that controls GPIOs
ftdi-gpio: Command Line Interface to test things out.

Currently the default setting is for my board, it has a custom product ID of 8350, I don't remember what the default product ID is bu it would be good to change the default settings to it (I think it's 0603??)

The 'interface' value specifies the 'bank', since the FT2232H has two 'banks' and the GPIOs for my device is on the 2nd bank then I put the default interface value as 2

The CLI looks like this:

``` bash
usage: ftdi-gpio [-h] [--vendor VENDOR] [--product PRODUCT]
                 [--interface INTERFACE] [-b WRBIT] [-m WRMASK] [-t RDBIT]
                 [-s DIRMASK] [-v VALUE] [-d]

Read and Write GPIOs with FTDI chip
Default Behavior is to read the GPIOs from the specified interface
usage: ftdi-gpio [options]

optional arguments:
  -h, --help            show this help message and exit
  --vendor VENDOR       Specify USB Vendor ID (Default: 0403)
  --product PRODUCT     Specify USB Product ID (Default: 8530)
  --interface INTERFACE
                        Specify USB Interface (Default: 2)
  -b WRBIT, --wrbit WRBIT
                        Specify the bit to set (Example: 6)
  -m WRMASK, --wrmask WRMASK
                        Specify a bitmask value to write (Example: 0x40)
  -t RDBIT, --rdbit RDBIT
                        Read back the mask and compare it with index (Example:
                        40)
  -s DIRMASK, --dirmask DIRMASK
                        Configure bit mask of the chip, will be default the
                        values associated with output
  -v VALUE, --value VALUE
                        For write set this to 1 or 0 for high or low
  -d, --debug           Enable Debug Messages

Examples:
	Something
```

**Examples**

Example: Reading all the GPIO for the lower section of the second bank:

``` bash
$ ./ftdi-gpio 
Pins: 0xFF
```

Example: Reading the value of one pin (bit 6), specifically a bit attached to a button:

```bash
Not Pressed (Button is high when not pressed)

$ ./ftdi-gpio --rdbit 6
Pins 6: True
Pins: 0xFF

Pressed
$ ./ftdi-gpio --rdbit 6
Pins 6: False
Pins: 0xBF
```

Example: Writing a one to another pin (bit 4), specifically the signal to reprogram my FPGA

```bash
Set GPIO 4 to an output and set that value high
$ ./ftdi-gpio --wrbit 4 --value 1
Pins: 0xFF

Set GPIO 4 to an output and set that output low
$ ./ftdi-gpio --wrbit 4 --value 0
Pins: 0xEF
```

Example: Explicitly declaring vendor id, product id and interface and then reading all the GPIOs

```bash
$ ./ftdi-gpio --vendor 0403 --product 8530 --interface 2
Pins: 0xFF
```

eblot, 2016-04-08:
Sorry, I had no HW to test GPIO for a while (and provide examples) but I should definitely work on this long-lasting issue.

eblot, 2016-04-13:
The latest version of PyFtdi (0.13.2) comes with GPIO examples. Hope this help.

closed 2016-04-13 by eblot