eblot/pyftdi #29: Pyinstaller, no usb backend

jksfiftyfive, 2015-08-31:
I have a GUI that Id like to turn into Windows 7/64bit EXE with Pyinstaller.

This GUI communicates to a device via an FTDI SPI controller. Ive used pyftdi to implement communication. GUI works great.

However, Im getting the "no backend available" when I try to run my EXE.

It boils down to this part of the code, which ive just shortened to include my candidate (\pyftdi\pyftdi\usbtools.py):

c = 'libusb0'
print 'usb candidate: ',c
m = getattr(um, c)
print 'm attr: ',m
backend = m.get_backend()
Running this in python works great. Running it in EXE, backend returns None.

I noticed that Pyinstaller does have hooks for matplotlib backend and usb hook, so i was hoping that my solution would be a combination of these two.

hooks/usb hook.py just finds the library, but does none of this backend stuff:

# NOTE: Mind updating run-time hook when adding further libs.
libusb_candidates = (
    # libusb10
    'usb-1.0', 'usb', 'libusb-1.0',
    # libusb01
    'usb-0.1', 'libusb0',
    # openusb
    'openusb',
)

I do notice that the libusb0.dll is copied to my EXE directory.

I realize this is more of a PyInstaller question, but figured I would post here as well

So my questions are: 1)How can i get Pyinstaller to either copy the usb-backend files that contain what its searching for, but cant seem to find? or just use the in-place structure that regular python seems to have no problem using

2)Does this have anything to do with the dll being local to my EXE dir, and do i need to copy some other files? (I did try just moving usb/backend dir to my EXE dir... no luck)

3)Can this be solved by adding more to usb hook to try to mimic the matlab-backend hook?
thanks

eblot, 2015-09-28:
I'm not really into Windows, but I guess you could ask for support on how to package libusb into a pyusb package on the pyusb ML.

eblot, 2015-09-28:
If you can provide a patch to ease support on Windows, I'd happy to merge it into pyftdi.

eblot, 2017-08-30:
Latest version provides basic installation instruction on Windows hosts.

It is probably still non trivial to package libusb into an python installer script, but it is nevertheless out-of-scope of this project. Sorry for that, I guess it might could be asked in the pyusb user support mailing list for example.

closed 2017-08-30 by eblot

ajsetter, 2021-05-06:
I'm not quite sure why this didn't work in OP's scenario, but I had this same issue and was able to resolve it by copying `C:\Windows\System32\libusb0.dll` to the root directory of the pyinstaller output, in the same directory as the main exe file.

This was with versions:

`pyftdi==0.52.9`
 `pyinstaller==4.3`

eblot, 2021-05-06:
Thanks for the feedback.

It is worth adding this solution to the troubleshooting guide then.