eblot/pyftdi #34: spi exchange() generates extra 8 clock cycles after read

BrettRD, 2016-05-28:
I've been using this lib with a FPGA dev board loaded with the ft2232HL.
The spi stuff works great, but there is an additional 8 clock cycles generated for every call to exhange().
The extra cycles happen at the very end of the exchange.
For example:
```
from pyftdi.spi import SpiController, SpiPort
spi = SpiController()
spi.configure(vendor=0x0403, product=0x6010, interface=1)
spiport = spi.get_port(1)
resp =  spiport.exchange([0x55, 0xAA, 0x5A], 3)
print str(resp)
spi.terminate()
exit(0)
```
This will generate 7\*8 clock cycles instead of 6\*8.
The additional cycles are even sent when the read count is zero.

This does not affect the number of bytes received by the computer, only the slave device.

eblot, 2016-05-28:
Hi BrettRD,

Try to tweak the SpiController() default parameter:
  spi = SpiController(turbo=False)
to see if it helps

BrettRD, 2016-05-28:
Hi elbot,
That's completely fixed it, Thanks!

eblot, 2016-05-28:
Good news! Thanks.

closed 2016-05-28 by eblot