eblot/pyftdi #37: UsbError: [Errno 13] Access denied (insufficient permissions)

manuelgodoy, 2016-11-06:
I am trying to interface with the FT232R. I am able to open the device successfully, but if I try to read data from it or write data to it I get this error:

"""python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyftdi/ftdi.py", line 685, in read_data
    return self.read_data_bytes(size).tostring()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyftdi/ftdi.py", line 674, in read_data_bytes
    raise FtdiError('UsbError: %s' % str(e))
pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)
"""

eblot, 2016-11-06:
Have you read & try: https://github.com/eblot/pyftdi#troubleshooting?

manuelgodoy, 2016-11-06:
I get this:

```
Can't create [-v].
(kernel) Kext com.apple.driver.AppleUSBFTDI not found for unload request.
Failed to unload com.apple.driver.AppleUSBFTDI - (libkern/kext) not found.
```

eblot, 2016-11-06:
`[something]` is the usual form in Unix word to say `something` is optional, while `<something>` means it is mandatory.

Therefore, you can either type

    sudo kextunload -bundle com.apple.driver.AppleUSBFTDI

or

    sudo kextunload -v -bundle com.apple.driver.AppleUSBFTDI

and read the man page for `kextunload`.

eblot, 2016-11-07:
It seems to use a usage issue, not a bug.

closed 2016-11-07 by eblot

manuelgodoy, 2017-07-12:
Hi @eblot,

Sorry for the confusion previously, but this is happening on my Raspberry Pi. Any idea how to fix this?

Thanks,

eblot, 2017-07-12:
It looks like a typical issue with udev:

You need to create a udev configuration file, for example:

    # /etc/udev/rules.d/11-ftdi.rules
    SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6011", GROUP="plugdev", MODE="0666"
    SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6010", GROUP="plugdev", MODE="0666"
    SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6001", GROUP="plugdev", MODE="0666"
    SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6014", GROUP="plugdev", MODE="0666"

Once this file has been created, unplug and plug the FTDI device in so that udev loads the new configuration from this file.

Be sure the user that wants to access the device is member of the `plugdev` group (or whatever you choose in the above configuration file).

    $ id

 If not, add it, then log out, log in:

    $ adduser $USER plugdev
    $ exit

eblot, 2017-07-12:
(I do not remember the product ID for FT232R, be sure to add it to the `11-ftdi.rules` file. Use `dmesg` or `lsusb` to check the USB product ID

manuelgodoy, 2017-07-12:
It is 6001.

This fixed the problem. Thank you very much.