eblot/pyftdi #38: Example request combining SPI and GPIO

chadkennedyonline, 2016-12-09:
I'm using the SpiController and SPI communications are working great. But I also need to toggle a reset line. I tried looking at gpio.py, but it's not obvious how I should initialize both SpiController and GpioController to the same FT232H, without conflicting "_ftdi" objects as well as conflicting pin directions.

So far, I've tried assigning the _ftdi object from one controller to the other, but I'm stuck on how to set direction for the GpioController without interfering with the direction for the SpiController, or vice-versa.

Could you please explain, or provide an example on how this could be done?

eblot, 2016-12-14:
FTDI chip have several, exclusive feature modes:
  * Asynchronous serial (UART), available on all FTDI chips
  * Bit-banging mode (GPIO), available on most, if not all, FTDI chips
  * Advanced serial mode (I2C, SPI, JTAG, ...), which use the MPSSE mode, only available on high-end FTDI chips (232x, 2232d/h, 4432d/h, ...)

`pyftdi` supports those three modes, although only SPI mode in implemented in MPSSE mode - I hope to find some spare time to add I2C and fix JTAG...

Each feature mode (UART/GPIO/MPSSE) is exclusive for a port (1 on 232, 2 on 2232, 4 on 4432 series). You can select GPIO mode on one port, and SPI on another port for example. You cannot mix both feature modes on the same port.

However, it is possible to use the unused pins of a SPI-enabled port as GPIOs, **but** the command set has nothing to do with the GPIO bit-band mode: to achieve GPIO management with a SPI-enabled port, MPSSE command set should be used whereas bit-bang command set is used with a GPIO-configured port. This is how /CS signal is driven when SPI mode is enabled for example.

Unfortunately, `pyftdi` does not offer MPSSE-GPIO support (but the /CS signal). It could be added though, but this requires some extra effort, especially dealing with exclusion of GPIO pins that are dedicated to /CS signalling.

The easiest workaround at the moment is to replace a 232h device with a 2232h device and dedicate one port to GPIO, the other one to MPSSE/SPI - or to provide a patch :-)

Leaving this ticket open as it would definitely be a nice feature to have.

chadkennedyonline, 2016-12-14:
Thanks for the response! I agree that your suggestion is the easiest workaround. In my case, I already had the Adafruit FT232H board on-hand and didn't want to wait for new hardware to come in. I don't have time to put together a clean patch at the moment, but I would like to share my workaround code here, just in case anyone needs to do the same thing I did. I'm using Windows 10 with Python 3.5. Your library works great on Windows btw, at least for my purposes.

```python
import time
from pyftdi.spi import SpiController
ctrl = SpiController(cs_count=1, turbo=True)
ctrl.configure(vendor=0x0403, product=0x6014, interface=1)  # Make sure you've loaded libusb-win32 using Zadig
spi = ctrl.get_port(cs=0)  # Chip select is 0 -- corresponds to D3
spi.set_frequency(3000000)

from array import array as Array
from pyftdi.ftdi import Ftdi
acbus_direction = 0x01 # Bit C0 is output for reset
ctrl._ftdi.write_data(Array('B', [Ftdi.SET_BITS_HIGH, 0x00, acbus_direction]))
time.sleep(1.0)
ctrl._ftdi.write_data(Array('B', [Ftdi.SET_BITS_HIGH, 0x01, acbus_direction]))
time.sleep(1.0)
```

Also, just for anyone interested, here's a list of links to some of the lower level documentation:
http://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232H.pdf
http://www.ftdichip.com/Support/Documents/AppNotes/AN_114_FTDI_Hi_Speed_USB_To_SPI_Example.pdf
http://www.ftdichip.com/Support/Documents/AppNotes/AN_135_MPSSE_Basics.pdf
http://www.ftdichip.com/Support/Documents/ProgramGuides/D2XX_Programmer's_Guide(FT_000071).pdf
http://www.ftdichip.com/Support/Documents/AppNotes/AN_108_Command_Processor_for_MPSSE_and_MCU_Host_Bus_Emulation_Modes.pdf

eblot, 2016-12-15:
> I'm using Windows 10 with Python 3.5. Your library works great on Windows btw, at least for my purposes.
This is great news.
If you had time to document the steps to make it work (dealing with libusb and so on), I would really appreciate it and merge it into the main documentation.

eblot, 2017-10-01:
FWIW, I've started implementing this feature. See spi-gpio branch.

nopeppermint, 2017-10-01:
nice, as I will need I2C and GPIO I will then have a good example how to combine, thanks!
but this could take some time..

eblot, 2017-10-04:
SPI w/ GPIO successfully tested with an OLED 0.96" display, where the SPI interface requires an extra GPIO to differentiate command from data requests (+ validated with a Saleae logic analysers as FTDI devices not always behave as expected :-)
Now delivered as v0.27.0

Closing this ticket as the original request is now available, feel free to open a new ticket for I2C ... or better, a pull request :-)

BTW, I've added a skeleton for MPSSE command tracer so that MPSSE commands & data are displayed in debug mode. It can be useful to track down issue. Only a small subset of MPSSE commands are implemented for now, I'll try to improve it with the next run.

closed 2017-10-04 by eblot

darlington72, 2018-02-26:
Dear all,

Finally, is it possible to control both SPI on AD0/1/2 and CS on AD4 + an extra gpio output on AC3 with an FT2232D? 
In my case, the SPI perfectly works with AD0/1/2 and CS on AD4 but I can't add the extra gpio output on AC3.

Thank you!