eblot/pyftdi #41: Support for FT4222H

gng1411, 2017-03-10:
Does this support FT4222H?

I quickly tried some commands and get the below errors:
>>> openedUSB = ctrl._ftdi.open_from_url('ftdi://ftdi:4222/1')
Traceback (most recent call last):
  File "<pyshell#215>", line 1, in <module>
    openedUSB = ctrl._ftdi.open_from_url('ftdi://ftdi:4222/1')
  File "C:\tools\python\lib\site-packages\pyftdi-0.22.0-py3.4.egg\pyftdi\ftdi.py", line 325, in open_from_url
    vendor, product, index, serial, interface = self.get_identifiers(url)
  File "C:\tools\python\lib\site-packages\pyftdi-0.22.0-py3.4.egg\pyftdi\ftdi.py", line 290, in get_identifiers
    cls.SCHEME, cls.VENDOR_IDS, cls.PRODUCT_IDS, cls.DEFAULT_VENDOR)
  File "C:\tools\python\lib\site-packages\pyftdi-0.22.0-py3.4.egg\pyftdi\usbtools.py", line 311, in parse_url
    raise UsbToolsError('Product ID 0x%04x not supported' % product)
pyftdi.usbtools.UsbToolsError: Product ID 0x107e not supported

OR ...
>>> ctrl._ftdi.open_mpsse(vendor=0x0403, product=0x601c)
Traceback (most recent call last):
  File "<pyshell#200>", line 1, in <module>
    ctrl._ftdi.open_mpsse(vendor=0x0403, product=0x601c)
  File "C:\tools\python\lib\site-packages\pyftdi-0.22.0-py3.4.egg\pyftdi\ftdi.py", line 364, in open_mpsse
    raise FtdiError('This device does not support MPSSE')
pyftdi.ftdi.FtdiError: This device does not support MPSSE

However, I did try below ...
>>> pyftdi.usbtools.UsbTools.get_device(vendor=0x0403, product=0x601c)
<DEVICE ID 0403:601c on Bus 000 Address 002>

(1)
Since I am using this (cool) library for the first time, maybe I am not doing it right. If so, let me know.

(2)
If this lib needs to be updated for FT4222H, since I have the device, I will be glad to contribute; however will need some hints on how to (update).

eblot, 2017-03-10:
Not yet :-)
I did not know it even existed.

I guess that SPI or I2C master could be easily adapted - except if FTDI changed their internal API.

However, it is the first FTDI chip I know to support the SPI slave mode, so I guess there would be more work to support this mode.

eblot, 2017-03-10:
The bad news is that the [MPSSE documentation](http://www.ftdichip.com/Support/Documents/AppNotes/AN_108_Command_Processor_for_MPSSE_and_MCU_Host_Bus_Emulation_Modes.pdf) does not refer to this device.

Other application notes only deal with their heavy APIs and libraries, not the FTDI chip opcodes :-(

gng1411, 2017-03-13:
Thank for the quick reply. :-)

eblot, 2017-07-06:
Closing as for now, documentation one FT4222H communication protocol is not available.

Reverse-engineering would probably be required, once again...

closed 2017-07-06 by eblot

Pe3ucTop, 2020-04-22:
Is there any news about support for FT4222H ?
It looks like there is official code examples now.
Very interested in QuadSPI it offer.

eblot, 2020-04-22:
As far as I can tell, FTDI have not published the FT4222H command set/doc/anything useful, but if you have some pointers I’ll be glad to have a look at them.

What they usually do is that they provide examples to use their close-source D2XX libraries, which are useless to implement an FTDI free concurrent API :-(

eblot, 2020-04-22:
BTW, looking @ the FT4222H datasheet, it seems to be a very different IP than the FT\*232\* series. So different that there would only be a little few pieces of code shared with PyFtdi. Said otherwise, this could not be part of PyFtdi, it is a different project, as it is a different IP.

FT\*232\* series use, to support I2C/SPI/JTAG serial protocol, the FTDI proprietary MPSSE protocol. This is a somewhat versatile protocol, but highly inefficient as the host needs to implement the serial bus protocol with low level MPSSE commands.

FT4222H on the other hand seems to implement the SPI and I2C protocols in HW, so the host would not have to implement them. This is likely far more efficient in term of bandwidth and even more in term of latency. A very different beast dedicated to SPI and I2C, whereas FT\*232\* are not.

To sum up: do not expect PyFtdi to ever support the FT4222H, although it could be nice to spawn another project dedicated to this specific device.