eblot/pyftdi #42: Very slow initialization on macOS Sierra (v10.12.3) - ~25s

LeonardMH, 2017-04-04:
I have just installed pyftdi today from pip. I am using Python 3.6 and pyftdi v0.22.0. I am seeing very slow initialization (ftdi.open) in my scripts.

I see this in the README

> Slow initialisation on OS X El Capitan
> It may take several seconds to open or enumerate FTDI devices.
> 
> If you run libusb <= v1.20.0, be sure to read the issue with OS X 10.11+.

I have to assume v1.20.0 is a typo since v1.0.21 is the latest version of libusb I currently see on the libusb website - [libusb SourceForge](https://sourceforge.net/projects/libusb/files/libusb-1.0/).

I have verified that I am running libusb-v1.0.21 (installed through homebrew).

```
▶ ls /usr/local/Cellar/libusb/
1.0.21
```

I initially set `DYLD_LIBRARY_PATH=/usr/local/lib` since that is where homebrew installs libraries. If I `ls` that directory I see the following libusb related libraries (I also have libusb-compat installed).

```
▶ ls /usr/local/lib | grep libusb
libusb-0.1.4.dylib
libusb-1.0.0.dylib
libusb-1.0.a
libusb-1.0.dylib
libusb.a
libusb.dylib
```

Since I thought PyFTDI might be using one of the wrong libusb .dylibs I changed my DYLD_LIBRARY_PATH to just directly include the library in the homebrew cellar: `DYLD_LIBRARY_PATH=/usr/local/Cellar/libusb/1.0.21/lib`.

In all of these cases I see that it takes a very long time to open the ftdi device. I'm connecting to an ft232h chip, I see that it is listed in `/dev` as:

```
▶ ls /dev | grep tty.usb
tty.usbserial-FTYPX4XC
```

So I then run the following script:

```python
from pyftdi.ftdi import Ftdi
import time

host = Ftdi()
ftdi_url = 'ftdi://ftdi:FTYPX4XC/1'
start_time = time.time()
host.open_from_url(ftdi_url)
end_time = time.time()

print("opened host in {}s".format(end_time - start_time))

import code
interpreter = code.InteractiveConsole(locals=locals())
interpreter.interact()
```

And I see that it prints `opened host in 25.21956515312195s`.

Is there something I'm doing wrong here? Any way to speed this up?

LeonardMH, 2017-04-04:
Using this url `'ftdi://ftdi:232h/1'` seems to help. I'm getting closer to 5 seconds to open a handle.

eblot, 2017-04-04:
> I have to assume v1.20.0 is a typo since v1.0.21 is the latest version of libusb I currently see on the libusb website 

Very true indeed.

> And I see that it prints opened host in 25.21956515312195s.

Weird. Using Sierra 10.12.4 (16E195):

    $> uname -a
    Darwin autagne.local 16.5.0 Darwin Kernel Version 16.5.0: Fri Mar  3 16:52:33 PST 2017; root:xnu-3789.51.2~3/RELEASE_X86_64 x86_64

    $> ls /dev | grep tty.usb
    tty.usbserial-141A
    tty.usbserial-141B

    # virtual environment creation
    $> pyvenv-3.5 ~/PyFtdi
    
    # installation from PyPi
    $> ~/PyFtdi/bin/pip3 install pyftdi
    Collecting pyftdi
      Downloading pyftdi-0.22.0-py35-none-any.whl (61kB)
        100% |████████████████████████████████| 71kB 522kB/s
    Collecting pyserial>=3.0 (from pyftdi)
      Downloading pyserial-3.3-py2.py3-none-any.whl (189kB)
        100% |████████████████████████████████| 194kB 1.3MB/s
    Collecting pyusb>=1.0.0 (from pyftdi)
      Downloading PyUSB-1.0.0.tar.gz (52kB)
        100% |████████████████████████████████| 61kB 958kB/s
    Installing collected packages: pyserial, pyusb, pyftdi
      Running setup.py install for pyusb ... done
    Successfully installed pyftdi-0.22.0 pyserial-3.3 pyusb-1.0.0
    
    # testing your script, with a single change (I do not have a FT232H at hand)
    # ftdi_url = 'ftdi://ftdi:2232h/1'  
    $> ~/PyFtdi/bin/python3.5 ./fttime.py
    opened host in 0.02749800682067871s
    Python 3.5.2 (default, Oct 13 2016, 22:50:05)
    [GCC 4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.38)] on darwin
    Type "help", "copyright", "credits" or "license" for more information.
    (InteractiveConsole)
    >>>

That is, 30 ms...

How many USB devices are connected to your Mac ?

It might be useful to enable `libusb` log messages to see if something obvious shows up:

    export LIBUSB_DEBUG=4

eblot, 2017-04-04:
Note that `ftdi://ftdi:FTYPX4XC/1` is not valid.

If you want to specify a FTDI device (in case several identical FTDI devices are connected at once), the serial number should be specified as the third argument: `ftdi://ftdi:232h:FTYPX4XC/1`

If you have a unique FTDI device, you can use a far shorter URL: `ftdi:///1`, although it is a bit slower:

    $> ~/PyFtdi/bin/python3.5 ./fttime.py ftdi:///1
    opened host in 0.1077120304107666s

    $> ~/PyFtdi/bin/python3.5 ./fttime.py ftdi://ftdi:2232h/1
    opened host in 0.023087024688720703s

LeonardMH, 2017-04-04:
Thanks for the quick response. For the rest of these tests I've changed my ftdi_url to be `ftdi://ftdi:232h`, where can I find documentation on the structure of the URLs?

Setting LIBUSB_DEBUG=4 shows that `libusb_get_device_list` is the offender. I can post the full log if needed, but here's the relevant part:

```
[ 0.047503] [00000e23] libusb: debug [darwin_event_thread_main] creating hotplug event source
[ 0.047735] [00000e23] libusb: debug [darwin_event_thread_main] darwin event thread ready to receive events
[ 0.047773] [00000307] libusb: debug [usbi_add_pollfd] add fd 3 events 1
[ 0.047863] [00000307] libusb: debug [libusb_get_device_list]
[ 5.049339] [00000307] libusb: debug [libusb_get_device_descriptor]
[ 5.049635] [00000307] libusb: debug [libusb_get_device_descriptor]
[ 5.049843] [00000307] libusb: debug [libusb_get_device_descriptor]
[ 5.049986] [00000307] libusb: debug [libusb_get_device_descriptor]
[ 5.050098] [00000307] libusb: debug [libusb_get_device_descriptor]
[ 5.050208] [00000307] libusb: debug [libusb_get_device_descriptor]
[ 5.050271] [00000307] libusb: debug [libusb_get_device_descriptor]
[ 5.050392] [00000307] libusb: debug [libusb_get_config_descriptor] index 0
[ 5.050489] [00000307] libusb: debug [libusb_get_config_descriptor] index 0
[ 5.050576] [00000307] libusb: debug [libusb_open] open 2.9
```

My MBP is connected to a Thunderbolt 3 dock with at least 4 usb devices attached. I've attached a screenshot of my system info USB device tree, if there is something that would be more useful in determining how many devices libusb is enumerating then let me know.

[image]

With that said, it doesn't seem to matter how many USB devices I have attached. I disconnected from my dock and just connected my FTDI cable through a USB-C-to-USB-A adapter and it still takes `libusb_get_device_list` 5 seconds to complete. It seems more like I'm hitting some sort of timeout than actually enumerating a bunch of devices.

eblot, 2017-04-04:
I'm afraid it looks like another issue with `libusb` and macOS kernel :-(
It might be interesting to contact the `libusb` team to check for any known issue.

I do not have a Mac with USB-C port (nor Thunderbolt bridge) to check if I can reproduce this issue.
On my MBPr (2013), `libusb` sequence looks like the following traces:

    [ 0.095300] [00000307] libusb: debug [usbi_add_pollfd] add fd 3 events 1
    [ 0.095336] [00000307] libusb: debug [libusb_get_device_list]
    [ 0.095391] [00000307] libusb: debug [libusb_get_device_descriptor]
    [ 0.095451] [00000307] libusb: debug [libusb_get_device_descriptor]

You can use USBProber to get additional USB information (macOS `lsusb` equivalent), which used to be an XCode additional developer tool. I'm not sure how to find it with the latest release(s) of XCode...

eblot, 2017-04-04:
> where can I find documentation on the structure of the URLs?

Ouch. I think I simply forgot to document it. To be done...

from `usbtools.py`:

     # general syntax: protocol://vendor:product[:index|:serial]/interface

That is:
  * protocol: always `ftdi`
  * vendor: the USB vendor ID of the manufacturer
     * ex: `ftdi` or `0x403`
  * product: the USB product ID of the device
     * ex: `232h` or `0x6014`
  * serial: the serial number as a string
  * index: an integer (no really useful, as it depends on the enumeration order on the USB buses
  * interface: the interface of FTDI device, starting from 1
    * e.g. `1` for 232*, `1` or `2` for 2232*, `1`..`4` for 4232* devices

All parameters but the interface are actually optional, PyFtdi tries to find the best match, _i.e._ if you have a single FTDI device connected to your system, `ftdi:///1` should be enough.

You can also asks PyFtdi to enumerate all the compatible devices with the special `ftdi:///?` syntax:

    $> ~/PyFtdi/bin/python3.5 ./fttime.py ftdi:///?
    Available interfaces:
      ftdi://ftdi:2232/1   (Dual RS232-HS)
      ftdi://ftdi:2232/2   (Dual RS232-HS)
    Please specify the USB device

LeonardMH, 2017-04-05:
Okay thanks, I will bring up this issue on the libusb repo and see if anyone there can help.

In the meantime, is there an API in PyFTDI that would allow me to connect to my device without calling `libusb_get_device_list`?

LeonardMH, 2017-04-05:
Well, this is weird, but this issue is fixed for me.

I downloaded the `libusb` sources last night just to take a look around. I noticed they have an example project `listdevs.c` which (among other things) calls `libusb_get_device_list`. Whenever I ran this example tool it worked very quickly, practically instantly. So I was back to thinking that PyFtdi was doing something wrong, maybe in the way `libusb_get_device_list` was being called, but I didn't have my FTDI cable to test against.

Testing again this morning I am again connected through my Thunderbolt hub to the FTDI cable. I ran the `listdevs` example in the libusb repository and it still worked very quickly. I then tried my python script using PyFTDI and it also worked quickly, connecting to the device in ~80ms.

The only thing I can think of that changed is that when I opened my Mac up last night it was rebooting itself, so I'm guessing one of the drivers was in a bad state and that was causing the issue.

closed 2017-04-05 by LeonardMH

eblot, 2017-04-05:
Good news! Thanks for the feedback.

themadsens, 2018-12-23:
Still have this issue with a freshly compiled libusb. However the 5 second delay can easily be cut down to 100ms. Detrimental effects unknown, buyers beware ..

```
--- a/libusb/os/darwin_usb.c
+++ b/libusb/os/darwin_usb.c
@@ -380,7 +380,7 @@ static void darwin_devices_detached (void *ptr, io_iterator_t rem_devices) {
 static void darwin_hotplug_poll (void)
 {
   /* not sure if 5 seconds will be too long/short but it should work ok */
-  mach_timespec_t timeout = {.tv_sec = 5, .tv_nsec = 0};
+  mach_timespec_t timeout = {.tv_sec = 0, .tv_nsec = 100000};

   /* since a kernel thread may nodify the IOInterators used for
    * hotplug notidication we can't just clear the iterators.
```

nicekwell, 2019-04-10:
> Still have this issue with a freshly compiled libusb. However the 5 second delay can easily be cut down to 100ms. Detrimental effects unknown, buyers beware ..
>
> ```
> --- a/libusb/os/darwin_usb.c
> +++ [...]

I have the same issue. I use stlink to program stm32. It base on libusb. But every time I would wait for 5 seconds on libusb_get_device_list()。
I solved this problem by update libusb from 1.0.22 to 1.0.23. The latest version on homebrew is 1.0.22. So we should compile from source code https://github.com/libusb/libusb

themadsens, 2019-04-10:
Yes, they since [merged the change](https://github.com/libusb/libusb/commit/0f7427777c2770b0b96a81021f85b6a0d1d2bfe4)