eblot/pyftdi #44: pip2 installs pyftdi version incompatible with python 2

sinback, 2017-05-04:
This feels like a silly place to report this issue, but I didn't see an easy way to contact the project maintainers on pypi.python.org.

I tried to install pyftdi for Python 2.7 to get some old code to work, but Pip2 grabbed pyftdi version 0.22.0, which doesn't support python2, rather than pyftdi 0.13.0, which as far as I can tell is the latest version that supports python 2. I don't maintain any python packages, but if it's straightforward to help pip2 and pip3 get different and correct versions of this library, that would save some time for people like me.

eblot, 2017-05-05:
Hi, thanks for reporting this issue.

It should be fixed with release 0.22.1 (51eefc0e87)

    $ virtualenv ~/Tmp/py27
    New python executable in /Users/eblot/Tmp/py27/bin/python2.7
    Also creating executable in /Users/eblot/Tmp/py27/bin/python
    Installing setuptools, pip, wheel...done.
    
    $ pyvenv ~/Tmp/py35
    
    $ ~/Tmp/py27/bin/pip2 install --no-cache-dir pyftdi
    Collecting pyftdi
      Downloading pyftdi-0.13.4.tar.gz (40kB)
        100% |████████████████████████████████| 51kB 2.7MB/s
    Collecting pyusb>=1.0.0b1 (from pyftdi)
      Downloading PyUSB-1.0.0.tar.gz (52kB)
        100% |████████████████████████████████| 61kB 70.6MB/s
    Collecting pyserial>=2.6 (from pyftdi)
      Downloading pyserial-3.3-py2.py3-none-any.whl (189kB)
        100% |████████████████████████████████| 194kB 23.0MB/s
    Requirement already satisfied: six in ./py27/lib/python2.7/site-packages (from pyftdi)
    Installing collected packages: pyusb, pyserial, pyftdi
      Running setup.py install for pyusb ... done
      Running setup.py install for pyftdi ... done
    Successfully installed pyftdi-0.13.4 pyserial-3.3 pyusb-1.0.0
    
    $ ~/Tmp/py35/bin/pip3 install --no-cache-dir pyftdi
    Collecting pyftdi
      Downloading pyftdi-0.22.1.tar.gz (48kB)
        100% |████████████████████████████████| 51kB 1.4MB/s
    Collecting pyusb>=1.0.0 (from pyftdi)
      Downloading PyUSB-1.0.0.tar.gz (52kB)
        100% |████████████████████████████████| 61kB 54.2MB/s
    Collecting pyserial>=3.0 (from pyftdi)
      Downloading pyserial-3.3-py2.py3-none-any.whl (189kB)
        100% |████████████████████████████████| 194kB 37.4MB/s
    Installing collected packages: pyusb, pyserial, pyftdi
      Running setup.py install for pyusb ... done
      Running setup.py install for pyftdi ... done
    Successfully installed pyftdi-0.22.1 pyserial-3.3 pyusb-1.0.0

Note that you can easily enforce a version at install time:

    pip2 install pyusb=0.13.4

closed 2017-05-05 by eblot

Renha, 2020-03-19:
Hello! Is that what suppose to happen?
```
C:\Python27\Scripts\pip install pyftdi
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no
 longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip
, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting pyftdi
  Using cached pyftdi-0.30.3.tar.gz (68 kB)
ERROR: Package 'pyftdi' requires a different Python: 2.7.17 not in '>=3.5'
```

eblot, 2020-03-19:
Python 2.7 should now longer be used.

You can force a very deprecated version of PyFtdi that supports Python2:

    pip2 install pyftdi==0.13.4

Although it is a bad idea ™

Renha, 2020-03-19:
yeah I know, but it's still being used. Thank you, the 0.13.4 version installs successfully!

eblot, 2020-03-19:
You're welcome.

Renha, 2020-03-23:
one more question: does 0.13.4 version support SpiGpioPort functionality?

eblot, 2020-03-23:
> one more question: does 0.13.4 version support SpiGpioPort functionality?
No, it is far too old, it has not been maintained for over 4 years.