eblot/pyftdi #45: Using single buffer with open-drain output for I2C SDAO

cioma, 2017-05-14:
This question is better suited for forum but I couldn't find a forum for pyftdi.

Some FTDI devices (FT232H) have open-drain capability on their outputs and others (FT2232H, FT4232H) don't. Absence of open-drain capability on I2C SCL prevents slave clock stretching support implementation. And absence of open-drain capability on I2C SDA prevents multi-master support implementation.

 that prevents I2C

SN74LVC1G07
Single
Buffer/Driver
With Open-Drain
Output

eblot, 2017-05-16:
Not that I know about, you are the first to ask :-)

rdpoor, 2018-01-14:
+1.  

For now, I've posted question / issue on StackOverflow.  

I have created a `pyftdi` tag and posted a question to https://stackoverflow.com/questions/48253119/pyftdi-finds-device-under-os-x-but-not-windows

eblot, 2020-12-31:
Long shot but Discussions are now available on GitHub. Closing this ticket as Discussions should fit this need.

closed 2020-12-31 by eblot