eblot/pyftdi #46: I2C with FT2232H?

cioma, 2017-05-15:
It seems I2C is not supported on FT2232H yet (pyftdi v0.22.1):
`pyftdi.ftdi.FtdiError: This device does not support drive-zero mode`

Is there a workaround for that?

eblot, 2017-05-16:
This is a HW limitation of the FTDI2232H.

I guess it could be leveraged at HW level: add 2 diodes on SCL and SDA so that FTDI can drive '0' level but not '1' level on the I2C bus, and some SW change to PyFtdi to ignore all "drive-zero" commands.

eblot, 2017-05-16:
Note: raising an error here is a bad idea, I'll change that.

cioma, 2017-05-16:
I suppose it should be possible to use an external buffer with open-drain output (e.g. TI SN74LVC1G07) on SCL and SDAO pins of FT2232H

eblot, 2017-05-16:
This would be another solution, yes.
A pure software solution would be to switch FTDI SDA line as input any time it should be high-Z, and switch it back to output "write" low level. This would make the code a bit more complex and less effective, but is probably not a big deal. I did not take time to experiment this mode.

vloewis, 2017-05-26:
I just had the same problem with a FT4232H.
I worked around it by catching the exception and setting a _zero_drive flag.
In the rest of the code i use a self._clock_low_data_z command instead of self._clock_low_data_high
whenever a self._read_bit or self._read_byte, follows, then i restore to self._clock_low_data_high.

Ugly, but works for me so far. (Reading/writing 24C32)

eblot, 2017-07-24:
Implemented in v0.25 release.

closed 2017-07-24 by eblot

alec178, 2018-02-22:
I met a pyftdi I2C problem with FT2232H.

When I use pyftdi I2cPort.read(16), SCL timing interval between each read byte is too closed. (see below waveform)
Is there any way to enlarge SCL timing interval of each byte?

[image]