eblot/pyftdi #47: Extra 0xff byte on SPI when using cs other than 0

vloewis, 2017-05-17:
Hello!
First of all thanks for this great lib.
I'm using it to interface a FPGA via a FTDI4232 SPI on interface 2. It works
great, however if i use CS 1 or 2 i see an extra 8 1 Bits clocked in before CS goes up.
It is not a real problem, because i can just ignore that byte. But i'm wondering 
nevertheless...
I've checked the pyftdi.spi code to the limited extend of my capabilities. 
This is what i believe it does, when writing:
- A byte sequence containing the MPSSE codes and the data is assembled
- The sequence starts with setting SCLK, MOSI and the selected CS to low (line 189)
- Next comes the data transmission command including data (line 236)
- If turbo == True and complete == True and silent_clock == False, the sequence finishes with 
-- ftdi.WRITE_BITS_TMS_NVE (line 153)
-- pulling up all outputs (line 155)

I am wondering wheter line 153 might be a problem. I believe it pulls up the 
TMS/CS-pin for eight clock cycles, while still emitting said clock cycles, however
my application does not use the default CS, which would be 0, but another one,
and thus clocks out eight additional bits.

-----------------------
I've commented line 153 and the eight bits are gone.

eblot, 2017-05-18:
Well, this is an old piece of code that definitely needs some improvement, I hope there is a better solution than the current implementation. Let me check if I can fix this properly.

eblot, 2017-05-19:
Could you give a try to `spi-cs-fix` branch?

vloewis, 2017-05-22:
> Could you give a try to spi-cs-fix branch?

I have no idea how to do that.

vloewis, 2017-05-22:
Somehow i closed this. Not my intention...
I'll give it a shot.

vloewis, 2017-05-23:
I've rewritten pyftdi.spi
- plus modes
- plus duplex
Do you mind if i send it to you? If not whereto?

eblot, 2017-06-06:
The best solution would be to create a pull request, or to attach a patch file.

eblot, 2017-06-10:
Extra useless bytes should have been fixed with v0.23.0 (a42e83b97)

eblot, 2017-06-12:
SPI mode 0, 1, 3 implemented.
If you known a workaround for mode 2, I would be interested with your implementation.

vloewis, 2017-06-12:
Would it be ok if i send it to emmanuel.blot@free.fr?

eblot, 2017-06-12:
If it is easier for you, yes, go ahead. Thanks.

vloewis, 2017-06-13:
i've send you the code. Unfortunately our EMail-Provider is pretty much down due to DDos, please tell me wheter you got the Mail

closed 2017-06-13 by vloewis