eblot/pyftdi #48: jtag  configure problem

RiccardoFrancescato, 2017-06-03:
running the test script for jtag, python reports a problem:
`Traceback (most recent call last):
  File "test.py", line 17, in setUp
    self.jtag.configure(vendor=0x0403, product=0x6011, interface=0)
  File "/usr/local/lib/python3.5/dist-packages/pyftdi/jtag.py", line 426, in configure
    interface=interface)
TypeError: configure() got an unexpected keyword argument 'vendor'`
how can i solve this?

RiccardoFrancescato, 2017-06-05:
i have found the problem! on jtag.py the connection is with the url  (line 200) instead of the normal version

ajaymills, 2017-06-06:
Hi, I understand that there are two "configure" functions in jtag.py, but then what needs to be done to make test.py run without error?

RiccardoFrancescato, 2017-06-06:
the problem is that in jtag.py the configure call open_mpsse_from_url instead of using the non url version
`def configure(self, url):
        """Configure the FTDI interface as a JTAG controller"""
        self._ftdi.open_mpsse_from_url(
            url, direction=self.direction, frequency=self._frequency)
        # FTDI requires to initialize all GPIOs before MPSSE kicks in
        cmd = Array('B', (Ftdi.SET_BITS_LOW, 0x0, self.direction))
        self._ftdi.write_data(cmd)
`

eblot, 2017-06-06:
JTAG implementation is quite outdated and has not been updated for a long time, as the JTAG engine itself would require a major rework.

ajaymills, 2017-06-06:
Could you elaborate on that a bit more? If it's just a matter of not being
able to use the URL-based interface, that's no big deal...Can I go back to
a previous version and use that?

RiccardoFrancescato, 2017-06-07:
The problem is not only the url-based interf. but also others pices of code that are obsolete or not tested.
@eblot i'm interested to develop the jtag platform (i have to use it for a master thesis work), can you send me some documentation for make jtag.py working well? 
thanks!

eblot, 2017-06-07:
jtag.py has been developed something like 8 years ago to drive a boundary scan on a custom ARM chip. It has not really been updated nor maintained since this early inception.

The main issue with with the TMS/TDI signals vs. TCK, which is likely bugged at some point. As I have no time to work with JTAG for over 5 years, it has not been fixed. If you want to take ownership of this topic, it would be very nice. Fixing URL vs regular MPSSE configuration is trivial (see spi.py and i2c.py for details), but fixing the signal management from the TAP engine is likely more difficult. You definitely need to understand how these signals work and their mutual dependencies, either from JTAG regular documentation or from open source implementation such as openocd.

RiccardoFrancescato, 2017-06-08:
@eblot thanks for the response, since i have to build a general jtag python interface, i would improve your code. For what concern the url vs regular i've already solved, i'm trying to figure out how the tms and tdi vs tck is built in the existing code and how to improve.
I have a question about the meanings of the parameters of cmd=Array('B'....)  parameters like `(Ftdi.RW_BYTES_PVE_NVE_LSB, blen, (blen >> 8) & 0xff)`
or `(Ftdi.RW_BITS_PVE_NVE_LSB, bit_count-1)`
thanks
RF

RiccardoFrancescato, 2017-06-08:
[image]
Asc can be seen from the picture there i noticed that the clock is not regular as it would be in the sense that after 8 signals there is a pause that obviously interfere with the decoding.
After some trials i have noticed that this problem came out from the division in bytes of the read/write, i was wandering if there is a way of setting a stream of n bits instead.

eblot, 2017-06-11:
I'm not sure to fully understand your comment:
There is not need for the JTAG clock to remain constant. The problem (AFAIR) is that the TDO signal may be clocked out one clock period too early or too late (sorry, it's been 8 years :-)).
There are two kind of FTDI MPSSE instructions: either bits or bytes. 'Bit'-based instructions can R/W from 0 to 7 bits at a time, while 'byte'-based instruction can R/W 1 up to 65536 bytes at a time.
Again, there should be no issue stopping/resuming the clock at any time, the problem is to output the proper bits/bytes value in sync with the clock.

BTW, thanks for letting me discover the Saleae Logic products, I just ordered one :-)

RiccardoFrancescato, 2017-06-13:
hi i have checked the jtag spec. and the clock signal can be interrupted.
so the problem is not the interruption.
after some trials i have noticed another thing, the idle of the clock is low and the idle of the tdo is hi; using a jtag interface (jlink) that works correctly i have noticed that the idle of the tck is hig and the idle of tdo is low.
is there a way to set the idle of clock to hy the idle of tdo to low and write on the rising edge and read on the falling one?

RiccardoFrancescato, 2017-06-15:
@eblot i'm doing some testing and i'm noticing that the lib is working fine (except some python errors that i'm solving).
At the moment i have changed the target board and the problems that i had before disappeared so i think that the problem was the board or the program inside the chip that locks some jtag functions
if you have suggestion for improving the lib  i'm here

closed 2017-06-26 by RiccardoFrancescato