eblot/pyftdi #53: UsbError on Ftdi.close()

gmarull, 2017-07-05:
Hi,

I am using the FTDI 232H as a GPIO controller. The device operates as expected, however, when my script finishes, the GpioController destructor raises an exception like this:

``
Exception ignored in: <bound method GpioController.__del__ of <pyftdi.gpio.GpioController object at 0x7fa2e0618128>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/pyftdi/gpio.py", line 48, in __del__
  File "/usr/local/lib/python3.5/dist-packages/pyftdi/gpio.py", line 75, in close
  File "/usr/local/lib/python3.5/dist-packages/pyftdi/ftdi.py", line 348, in close
  File "/usr/local/lib/python3.5/dist-packages/pyftdi/ftdi.py", line 575, in set_latency_timer
  File "/usr/local/lib/python3.5/dist-packages/pyftdi/ftdi.py", line 914, in _ctrl_transfer_out
pyftdi.ftdi.FtdiError: UsbError: [Errno None] Other error
``

I have tried commenting out the `set_latency_timer` call in the `Ftdi.close()` function and then, as expected, no exception is raised and scripts terminates successfully.

Have anybody else experienced this problem?

Thanks,
-Gerard

eblot, 2017-07-06:
I think this comes from an old, bad idea:

`close()` is called automatically when the Python VM is about to destroy the instance, which may occur before **or** after the USB connection is actually closed. This may lead to call the underlying `pyusb` module (and `libusb`) on an already-closed connection, hence the error.

The `__del__()` has been removed from other PyFtdi module a while ago, but are still present on GPIO and JTAG submodules. I will remove them.

Either call close() explicitly, or leave pyusb/libusb to handle the situation when the Python interpreter is shut down.

eblot, 2017-07-06:
Fixed in 8f671a9.

closed 2017-07-06 by eblot