eblot/pyftdi #60: Accessing GPIO on a device with custom PID

nickvanderstel, 2017-08-04:
I see that the Ftdi class has a method to add a device with a custom PID, but how can I use the GpioController class with this device? I have a FT2232H but the PID is 0x8a98 so I'm getting the following error:
```
ERROR: test_gpio (__main__.GpioTestCase)
Simple test to demonstrate bit-banging.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "gpio.py", line 98, in test_gpio
    gpio.open(mask)
  File "gpio.py", line 47, in open
    self._gpio.open_from_url('ftdi://0x0403:0x8a98/1', direction=out_pins)
  File "/usr/lib/python3.6/site-packages/pyftdi/gpio.py", line 75, in open_from_url
    ftdi.open_bitbang_from_url(url, direction=direction, **kwargs)
  File "/usr/lib/python3.6/site-packages/pyftdi/ftdi.py", line 507, in open_bitbang_from_url
    vendor, product, index, serial, interface = self.get_identifiers(url)
  File "/usr/lib/python3.6/site-packages/pyftdi/ftdi.py", line 305, in get_identifiers
    cls.SCHEME, cls.VENDOR_IDS, cls.PRODUCT_IDS, cls.DEFAULT_VENDOR)
  File "/usr/lib/python3.6/site-packages/pyftdi/usbtools.py", line 359, in parse_url
    raise UsbToolsError('Product ID 0x%04x not supported' % product)
pyftdi.usbtools.UsbToolsError: Product ID 0x8a98 not supported
```

eblot, 2017-08-07:
You need to add the following piece of code:

    from pyftdi.ftdi import Ftdi
    Ftdi.add_custom_product(Ftdi.DEFAULT_VENDOR, 0x9999)

before the first call to

    GpioController().open_from_url(url, ...)

I should probably add this kind of example to the PyFtdi documentation

nickvanderstel, 2017-08-08:
That did the trick. Thanks!

closed 2017-08-08 by nickvanderstel

eblot, 2017-08-09:
Documentation have been updated in v0.25.3 to make this customization API more explicit.