eblot/pyftdi #61: FTDI maximum communication speed

ManuK42, 2017-08-07:
In my setup I communicate via the C232HM FTDI cable with an ADXL345 accelerometer. I ran some tests to compare the communication speed with my former setup (GPIO Header of a RaspberryPi).

I read 20 twenty data sets 1000 times and documented the mean value. One data set contains 7 Bytes (1 to set the register + 6 read bytes). These are the results:

SPI Frequency | Time per data set via GPIO SPI | The per data set via FTDI SPI
4 MHz | 1.47 ms | 4.95 ms
2 MHz | 2.13 ms | 5.80 ms
1 MHz | 2.84 ms | 5.49 ms
0.5 MHz | 4.87 ms | 5.52 ms

It seems that higher SPI speeds do not really affect the measured time via FTDI, so I assume that either the communication between my PC and the FTDI device or the processing through different python libraries is the limiting factor. 
 
Do you have any experience with that? Are there any known speed limits or methods to improve the communication speed via FTDI?
If not, do you believe I could improve it by using the C++ API and D2XX driver which is provided by FTDI?
 
Thank you very much for your time.

eblot, 2017-08-07:
I'm not really surprised with these results.
There are a lot of layers between PyFtdi and the actual SPI slave, at least (but not limited to)
 - pyftdi
 - pyusb
 - libusb
 - syscall
 - kernel usb stack
 - usb EHCI (host controller)
 - usb bus
 - FTDI usb I/F
 - FTDI MPSSE engine
 - SPI bus

USB bulk packets can be sent every 1ms with full speed device, and every 125µs on a high-speed device. The way PyFTDI works is that it generates a sequence of commands for the MPSSE engine to execute, sends the command sequence into a USB frame, and expects a data sequence back from the FTDI device. It is likely that the full round trip takes several 125µs micro frame (assuming a high speed USB connection):
 - 1 micro frame to send the command & data
 - 1 or more micro frames that cannot be used as the MPSSE executes (and fills its FIFO)
 - 1 micro frame to receive the data
Moreover, the whole SW stack & Python interpreter is likely to take a couple of other USB slots to traverse.

The end result is that the overhead for very short data packets (such the one you use: 1+6 bytes) is likely to be far more important than the actual data transfer over the SPI bus.

If you look at the pyspiflash companion project, you can see that the maximum readout bandwidth is about 1.3Mbyte/s, that is roughly ~11 MBps, with SPI chunks that are as large as possible. Increasing the SPI clock would not help. 

I guess the Python interpreter may slow down a little bit the whole system, but I'm not sure it is the major culprit, especially with nowadays PCs

To increase throughput (assuming you cannot change/optimise the SPI slave protocol...), I guess the most efficient solution would be to prepare several SPI commands into a single USB buffer, so that the  MPSSE engine executes several slave requests out of the same USB frame. The MPSSE engine would then put several SPI slave responses into the same USB frame back to the host. The host would have to split out the received frame into the proper output responses. PyFtdi is not actually designed to support this, but I think it would not be really complex to do.

In other words, the current implementation works this way:
  - push "write to SPI" command to the output buffer
  - push "data to write" to the output buffer ("set to register" in your example)
  - push "read from SPI" command to the output buffer (6 bytes in your example)
  - push "read immediate" command to the output buffer (so that the SPI data are sent back to the host as soon as possible, which decrease latency but also decrease the throughput)
  - send the output buffer
  - receive the response buffer
  - remove FTDI header

A more efficient implementation would work this way:
  - repeat N times:
     - push "write to SPI" command to the output buffer
     - push "data to write" to the output buffer
     - push "read from SPI" command to the output buffer
  - push "read immediate" command to the output buffer
  - send the output buffer
  - receive the response buffer
  - remove FTDI header
  - (split out the receive buffer into N response)
where N is an integer number computed so that a integral number of commands/data fits into a USB frame

It could be implemented with a new API to split the generation of the output buffer from the actual transmission over the USB bus. It would definitely increase the throughput. The "request" buffer could even be generated once, and as soon as the response buffer is received.

Something interesting to have a look at :-) 

I'm not sure C++/D2XX would make a large difference, I think it is more about the content of the USB buffers rather than how you generate them, I believe, but I might be wrong.

ManuK42, 2017-08-09:
Wow, thank you very much for this detailed answer. I just changed the data package size and the results are impressive: Reading 150 Bytes at once is about 7 times faster than reading 6 Bytes 25-times in a row.

eblot, 2017-08-09:
Great. You can tailor the actual packet size so that it fit within the FTDI FIFO. This should give you the best results, throughput-wise.

Checkout the fifo_sizes property, something

    spi = SpiController()
    # ...
    tx_size, rx_size = spi._ftdi.fifo_sizes

which should be 1KiB (1024 bytes) for a FT232H.

eblot, 2017-08-30:
Closing, but feel free to provide an additional API to prepare and concatenate requests before sending it over the USB bus :-)

closed 2017-08-30 by eblot