eblot/pyftdi #62: Accessing serial and GPIO simultaneously

nickvanderstel, 2017-08-08:
I have found that, on Linux, accessing FTDI GPIO causes the FTDI USB Serial Device to be disconnected:

`[24051.378487] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0`

If I have the FTDI serial port (pyserial) open before accessing the FTDI GPIO, I get IO errors in pyserial after accessing the GPIO. I also noticed that calling GpioController.close() does not reload the FTDI USB Serial Device.

Is there a way to access both simultaneously or is this a know limitation?

eblot, 2017-08-08:
I'm not sure to understand the exact HW configuration:

Do you attempt to use the same FTDI interface for both GPIO and serial communication, or to use GPIO on one interface and serial on another one?

FT232 devices have only one interface, FT2232H two, FT4232H four.

Linux kernel can either dedicated a FTDI interface to its serial driver (ftdi_sio) or to a user space application (through libusb/pyusb/pyftdi), but not at once.

PyFtdi also supports one mode per interface (one of: serial, GPIO, SPI, I2C, JTAG). There is a request to support (SPI or I2C or JTAG) and GPIO at once [see #38], but so far it is not implemented.

I'm not sure whether a FTDI can support UART and GPIO on the same interface (I do not think so, but I might be wrong).

I have not tested the mixed kernel space/use space mode (i.e. let the kernel manage a serial interface while managing another interface with PyFtdi), but I would be surprised if it worked out of the box, as PyFtdi attempts to claim all the interfaces from a device.

However it should be possible to use two interfaces (for example serial and GPIO) from user space with PyFtdi. For example, with an FT2232H.
  * run one session the ``pyterm.py`` test application with URL ftdi:///1
  * run another session of ``pyterm.py`` with URL ftdi:///2
If you connect ADBUS0 to BDBUS1 and ADBUS1 to BDBUS0, you should see the character you type in one console appear in the other console.

I have not tried to open both ports from the same Python interpreter for a while, but it should work (I should add a unit test for this use case BTW...)

You do not need ``pyserial`` (and ``serialext``) to use PyFtdi as a UART driver, but it would require to re-implement some part of the ``pyserial`` to achieve the same feature set. You should not need ``pyserial`` if you do not plan to use the UART mode of your FTDI device.

nickvanderstel, 2017-08-08:
I was trying to access serial on the wrong interface - so I was trying to use the same interface for both GPIO and serial. I corrected that and have both working now. Thanks!

closed 2017-08-08 by nickvanderstel

eblot, 2017-08-09:
> (I should add a unit test for this use case BTW...)

Done with v0.25.3

shuffle2, 2017-11-30:
I have a FT232H-based cable (C232HM-DDHSL-0) that i want to use UART and GPIO simultaneously with.

According to FT232H manual, this should be possible:
> The ACBUS pins can also be individually configured as GPIO pins, similar to asynchronous bit bang mode.
It is possible to use this mode while the UART interface is being used, thus providing up to 4 general
purpose I/O pins which are available during normal operation. 

I haven't looked at making it work with pyftdi yet. Any comments?

pazucj, 2018-02-08:
U should write on terminal:
sudo modprobe ftdi_sio
sudo modprobe usbserial
 but if u want to disable VCP and enable D2XX u should write:
sudo rmmod ftdi_sio
sudo rmmod usbserial