eblot/pyftdi #63: index usage with Windows

frekr1, 2017-08-09:
I've noticed that you changed since a while the way usbtools is collecting usb devices using python "set" in place of "list". Unfortunately in Windows, the same FTDI device is visible through libusb-win32 as multiple devices in particular:
in case of an FTDI device with 2 ports: 3 usb devices are visible (1 with 2 interfaces and 2 with 1 interface)
in case of an FTDI device with 4 ports: 5 usb devices are visible (1 with 4 interfaces and 4 with 1 interface)

unfortunately python "set" are saved with no order so it is not possible once collected in the same set, to distinguish the device that has multiple interfaces while using list the first one in the list (means with index=0) is always the one with multiple interfaces. Could you consider a switch back into the old style list kind of usb device collection a non disruptive proposal?

eblot, 2017-08-09:
I did not think anyone would be select a device with the index, as the index depends on how the USB is enumerated, and the order the devices are plugged in. 

Is there a reason why you cannot select the device with its serial number?

frekr1, 2017-08-10:
Indeed I'm using the serial number but since the 3 or 5 devices shows the same serial that's why I need to use index to discriminate the one with 2 interfaces. Here an example using a FT2232HL that features 2 ports launching the Test Win program available with Libusb for win32:

```
bus-0/\\.\libusb0-0006--0x0403-0x6010     0403/6010
- Manufacturer : FTDI
- Product      : USB <-> Serial Converter
- Serial Number: 16390001
bLength:             18
bDescriptorType:     01h
bcdUSB:              0200h
bDeviceClass:        00h
bDeviceSubClass:     00h
bDeviceProtocol:     00h
bMaxPacketSize0:     40h
idVendor:            0403h
idProduct:           6010h
bcdDevice:           0700h
iManufacturer:       1
iProduct:            2
iSerialNumber:       3
bNumConfigurations:  1
  wTotalLength:         55
  bNumInterfaces:       2
  bConfigurationValue:  1
  iConfiguration:       0
  bmAttributes:         80h
  MaxPower:             50
    bInterfaceNumber:   0
    bAlternateSetting:  0
    bNumEndpoints:      2
    bInterfaceClass:    255
    bInterfaceSubClass: 255
    bInterfaceProtocol: 255
    iInterface:         2
      bEndpointAddress: 81h
      bmAttributes:     02h
      wMaxPacketSize:   512
      bInterval:        0
      bRefresh:         0
      bSynchAddress:    0
      bEndpointAddress: 02h
      bmAttributes:     02h
      wMaxPacketSize:   512
      bInterval:        0
      bRefresh:         0
      bSynchAddress:    0
    bInterfaceNumber:   1
    bAlternateSetting:  0
    bNumEndpoints:      2
    bInterfaceClass:    255
    bInterfaceSubClass: 255
    bInterfaceProtocol: 255
    iInterface:         2
      bEndpointAddress: 83h
      bmAttributes:     02h
      wMaxPacketSize:   512
      bInterval:        0
      bRefresh:         0
      bSynchAddress:    0
      bEndpointAddress: 04h
      bmAttributes:     02h
      wMaxPacketSize:   512
      bInterval:        0
      bRefresh:         0
      bSynchAddress:    0
bus-0/\\.\libusb0-0008--0x0403-0x6010     0403/6010
- Manufacturer : FTDI
- Product      : USB <-> Serial Converter
- Serial Number: 16390001
bLength:             18
bDescriptorType:     01h
bcdUSB:              0200h
bDeviceClass:        00h
bDeviceSubClass:     00h
bDeviceProtocol:     00h
bMaxPacketSize0:     40h
idVendor:            0403h
idProduct:           6010h
bcdDevice:           0700h
iManufacturer:       1
iProduct:            2
iSerialNumber:       3
bNumConfigurations:  1
  wTotalLength:         32
  bNumInterfaces:       1
  bConfigurationValue:  1
  iConfiguration:       0
  bmAttributes:         80h
  MaxPower:             50
    bInterfaceNumber:   0
    bAlternateSetting:  0
    bNumEndpoints:      2
    bInterfaceClass:    255
    bInterfaceSubClass: 255
    bInterfaceProtocol: 255
    iInterface:         2
      bEndpointAddress: 81h
      bmAttributes:     02h
      wMaxPacketSize:   512
      bInterval:        0
      bRefresh:         0
      bSynchAddress:    0
      bEndpointAddress: 02h
      bmAttributes:     02h
      wMaxPacketSize:   512
      bInterval:        0
      bRefresh:         0
      bSynchAddress:    0
bus-0/\\.\libusb0-0009--0x0403-0x6010     0403/6010
- Manufacturer : FTDI
- Product      : USB <-> Serial Converter
- Serial Number: 16390001
bLength:             18
bDescriptorType:     01h
bcdUSB:              0200h
bDeviceClass:        00h
bDeviceSubClass:     00h
bDeviceProtocol:     00h
bMaxPacketSize0:     40h
idVendor:            0403h
idProduct:           6010h
bcdDevice:           0700h
iManufacturer:       1
iProduct:            2
iSerialNumber:       3
bNumConfigurations:  1
  wTotalLength:         32
  bNumInterfaces:       1
  bConfigurationValue:  1
  iConfiguration:       0
  bmAttributes:         80h
  MaxPower:             50
    bInterfaceNumber:   1
    bAlternateSetting:  0
    bNumEndpoints:      2
    bInterfaceClass:    255
    bInterfaceSubClass: 255
    bInterfaceProtocol: 255
    iInterface:         2
      bEndpointAddress: 83h
      bmAttributes:     02h
      wMaxPacketSize:   512
      bInterval:        0
      bRefresh:         0
      bSynchAddress:    0
      bEndpointAddress: 04h
      bmAttributes:     02h
      wMaxPacketSize:   512
      bInterval:        0
      bRefresh:         0
      bSynchAddress:    0	  

```

eblot, 2017-08-10:
Holly cr*ap :-) I wonder if this comes from just another Windows stupidity (™)  or libusb under Windows.

Could you tell me how to reproduce this issue (I barely use Windows), so I can try to understand better this? - or at least a pointer to a valid & up-to-date page on what/how to install the required stuff (libusb, I remember there were some kind of tool to create the proper configuration for libusb, ...) on Windows 7. 

Thanks.

frekr1, 2017-08-11:
First of all you need to install libusb-win32-devel-filter-1.2.6.0.exe. This is available form here:
[https://sourceforge.net/projects/libusb-win32/files/libusb-win32-releases/1.2.6.0/](url)
Before using it, you should install the so called "filter" so that libusb can coexist with specific peripherals usb drivers ... so then go into the Libusb-Win32 menu (it should be something like C:\ProgramData\Microsoft\Windows\Start Menu\Programs\LibUSB-Win32) and then into the folder Class Filter launch the installation of the filter for all the usb devices i.e. Install all class filters. Once done libusb can be used "in parallel" with original drivers. 
Then inside the same Libusb-Win32 root menu you should have also the "Test (Win) Program" That I used to provide you the example....That's it.

Similar thing can be see also if you use zadig and try install specific libusb drivers replacing the FTDI ones.  [http://zadig.akeo.ie/](url) just launching the executable zadig-2.3.exe and un-checking Options/Ignore Hubs or Composite Parents you should see from the drop-down list that for exampe for a 2 port FTDI ther are 3 items: 1 Composite Parent with 2 interfaces and 2 separated interfaces. 

Finally I'm quite sure all of this comes from Windows since it is independent from kind of libusb is used.
Let me know.

eblot, 2017-08-25:
OMG. Know I remember why I never supported this sh*tty "OS" :-(

Holly crap:

With both interfaces (port A and port B) declared to the filter:

    Z:\pyftdi>python pyftdi\serialext\tests\pyterm.py -p ftdi:///?
    Available interfaces:
      ftdi://ftdi:2232:FTVMFO20/1   (FT2232H-56Q MiniModule)

Windows only sees the first port.

    Z:\pyftdi>python pyftdi\serialext\tests\pyterm.py -p ftdi:///?
    Available interfaces:
      ftdi://ftdi:2232:FTVMFO20/1   (FT2232H-56Q MiniModule)
      ftdi://ftdi:2232:FTVMFO20/1   (FT2232H-56Q MiniModule)
      ftdi://ftdi:2232:FTVMFO20/2   (FT2232H-56Q MiniModule)

Windows sees three ports.

However, when the composite device is the only FTDI component declared to the filter, Windows looks less stupid (there is a very long way to go to make it usable, still):

    Z:\pyftdi>python pyftdi\serialext\tests\pyterm.py -p ftdi:///?
    Available interfaces:
      ftdi://ftdi:2232:FTVMFO20/1   (FT2232H-56Q MiniModule)
      ftdi://ftdi:2232:FTVMFO20/2   (FT2232H-56Q MiniModule)

However, whatever the configuration, communication with any port fails miserably:

======================================================================
ERROR: _test_uart_loopback (__main__.UartTestCase)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "Z:\pyftdi\pyftdi\serialext\protocol_ftdi.py", line 52, in open
    device = Ftdi.create_from_url(self.port)
  File "Z:\pyftdi\pyftdi\ftdi.py", line 294, in create_from_url
    device.open_from_url(url)
  File "Z:\pyftdi\pyftdi\ftdi.py", line 383, in open_from_url
    self.open(vendor, product, index, serial, interface)
  File "Z:\pyftdi\pyftdi\ftdi.py", line 408, in open
    self.usb_dev = UsbTools.get_device(vendor, product, index, serial)
  File "Z:\pyftdi\pyftdi\usbtools.py", line 158, in get_device
    if dev.get_active_configuration().bConfigurationValue != 1:
  File "C:\Users\WinIsCrap\AppData\Local\Programs\Python\Python35\lib\site-packages\usb\core.py", line 875, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "C:\Users\WinIsCrap\AppData\Local\Programs\Python\Python35\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\WinIsCrap\AppData\Local\Programs\Python\Python35\lib\site-packages\usb\core.py", line 239, in get_active_configuration
    bConfigurationValue=self.backend.get_configuration(self.handle)
  File "C:\Users\WinIsCrap\AppData\Local\Programs\Python\Python35\lib\site-packages\usb\backend\libusb0.py", line 510, in get_configuration
    100)
  File "C:\Users\WinIsCrap\AppData\Local\Programs\Python\Python35\lib\site-packages\usb\backend\libusb0.py", line 593, in ctrl_transfer
    timeout
  File "C:\Users\WinIsCrap\AppData\Local\Programs\Python\Python35\lib\site-packages\usb\backend\libusb0.py", line 431, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: The I/O operation has been aborted because of either a thread exit or an application request.\r\n\n'

eblot, 2017-08-25:
BTW, pyusb seems to use the very deprecated libusb 0.x... libusb-win32 is still using this old API?

eblot, 2017-08-25:
Could you give a try to the winkludge branch?

It works with one and two FTDI2232H device on "my" Windows 7 x64, and both UART tests are executing as expected (`uart.py` and `pyterm.py`). Thanks.

eblot, 2017-08-30:
Merge into the master branch, as it brings no regression to non-Windows hosts.

eblot, 2017-10-04:
No update. This fix has been delivered to the master branch, feel free to re-open if it does not address the original issue.

closed 2017-10-04 by eblot