eblot/pyftdi #67: No such device (it may have been disconnected)

manuelgodoy, 2017-08-24:
In an effort to make my script more robust, I am purposely disconnecting and connecting back my usb device. After doing this I get this error when I try to send any command to my Ftdi object:

```python
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 1314, in _ctrl_transfer_out
    array('B').frombytes(data), self.usb_write_timeout)
  File "/usr/local/lib/python3.6/site-packages/usb/core.py", line 1043, in ctrl_transfer
    self.__get_timeout(timeout))
  File "/usr/local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 883, in ctrl_transfer
    timeout))
  File "/usr/local/lib/python3.6/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 19] No such device (it may have been disconnected)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/EPR/src/application/controller/controller.py", line 19, in open
    super(Controller, self).open(vendor = vendor, product = product)
  File "/usr/local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 402, in open
    self.purge_buffers()
  File "/usr/local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 735, in purge_buffers
    self.purge_rx_buffer()
  File "/usr/local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 722, in purge_rx_buffer
    if self._ctrl_transfer_out(Ftdi.SIO_RESET, Ftdi.SIO_RESET_PURGE_RX):
  File "/usr/local/lib/python3.6/site-packages/pyftdi/ftdi.py", line 1316, in _ctrl_transfer_out
    raise FtdiError('UsbError: %s' % str(e))
pyftdi.ftdi.FtdiError: UsbError: [Errno 19] No such device (it may have been disconnected)
```

The only way I can bring my USB device back is to close my python process, restart it and reconnect to the Ftdi object. This is not very practical when I intend to deploy my device remotely without ssh access. Any suggestions on how to reconnect to the device within the python process? (I tried `close()` and then `open()` but no luck.

eblot, 2017-08-25:
Hi,

I guess I missing out some context. Using the Git version, I wrote a simple unit test to open, write, read back, close on the first UART port (TX and RX pins being connected):

    url = environ.get('FTDI_DEVICE', 'ftdi://ftdi:2232h/1')
    port = serial_for_url(url, baudrate=1000000)
    for cycle in range(10):
        print("cycle %d" % cycle)
        try:
            if not port.is_open:
                print("open")
                port.open()
            port.timeout = 1.0
            something_out = self.generate_bytes()
            port.write(something_out)
            something_in = port.read(len(something_out))
            self.assertEqual(something_in, something_out)
        finally:
            print("close")
            port.close()

It runs without any issue.

Does it work on your setup?

manuelgodoy, 2017-08-25:
Hi,

I meant physically disconnecting the device (I am using a raspberry Pi and an FT232R).

eblot, 2017-08-28:
You need to flush the device cache, as any time you plug a device on the USB bust, it is enumerated and may appear as a different USB device.

Example with UART:

    from pyftdi.usbtools import UsbTools
    from pyftdi.serialext import serial_for_url
    
    port = serial_for_url(url)
    try:
        while True:
            if not port.is_open:
                port.open()
            print(port)
            input("open, please disconnect")
            try:
                port.close()
            except Exception as ex:
                print(ex)
            input("closed, please reconnect")
            UsbTools.flush_cache()
    except:
        port.close()

eblot, 2017-08-30:
Feel free to re-open if it does not work for you

closed 2017-08-30 by eblot

manuelgodoy, 2017-09-05:
Not working for me.