eblot/pyftdi #70: Support selection by "USB path" in URL

axel-h, 2017-10-15:
Currently the URL scheme is `protocol://[vendor][:[product][:index|:serial]]/interface`. Can you add a way to use the connected USB port (ie the USB path along all hub ports) to select a device? I often have the use case that multiple devices are attached and I'm working with a specific USB port. 
USB Path could be: `<root hub>-<port>[.<port>[.<port>[...]]]`

Not sure is there is a nice way to add the path in the existing scheme:
- `protocol://[<usb-path>@][vendor][:[product][:index|:serial]]/interface`
- `protocol://[vendor][:[product][:index|:serial]]/interface[@<usb-path>]`

Furthermore, I've noticed that `get_identifiers('ftdi:///?')` does not list multiple devices if there is no serial (seem these are fake FTDI chip then anyway). There is only one entry for all devices. With the USB path feature from above the devices could be separated.

Axel

eblot, 2017-11-01:
I'm not sure if it is feasible through the PyUSB/libusb stack. I need to go through their respective documentation. I'm even less sure this could be portable across the 3/4 major OSes (Linux, macOS, Windows, *BSD)...
The usual workaround is to use the serial number - but it requires an EEPROM chip connected to each FTDI device.

I'm no sure about the issue with `ftdi:///?`. What do you mean by "no serial" here?

eblot, 2019-06-04:
No feedback, closing.

closed 2019-06-04 by eblot