eblot/pyftdi #71: UsbTools.find_all error: 'int' object is not iterable

holachek, 2017-10-31:
Hi, I'm trying to find my FT4232H Mini Module eval board with pyftdi from my Windows 10 machine.

From the interactive Python 3 console, I type
`import pyftdi
vid=0x0403
pid=0x6011
pyftdi.usbtools.UsbTools.find_all(vid, pid)`

To which I receive an error:
`[...]
for v, p, in vps:
TypeError: 'int' object is not iterable`

Is this a bug?

eblot, 2017-11-01:
You need to provide a sequence of *(vid,pid)* tuples:

    pyftdi.usbtools.UsbTools.find_all([(vid, pid)])

The documentation has a typo, it should be read as:

    def find_all(vps, nocache=False):
        """Find all devices that match the specified vendor/product pairs.

           :param vps: a sequence of 2-tuple (vid, pid) pairs
           :type vps: list(tuple(int, int))

holachek, 2017-11-01:
Works now. Thanks!

closed 2017-11-01 by holachek