eblot/pyftdi #72: I2C hold time after START seems to be too short

PetteriAimonen, 2017-11-11:
I'm currently seeing about 2.2 us time from when SDA goes low for START, to the time when SCL goes low. I'm running the bus at 100 kHz but the time seems to be the same at any speed.

[image]

However, the I2C bus specification requires this time (t HD;STA) to be atleast 4 us:

[image]

[image]

This seems to cause intermittent communication problems atleast with bq20z45 battery IC. What would be the easiest way to increase the delay?

eblot, 2017-11-27:
Sorry for the delay.

Try to change

   self._start = data_low*4 + clock_low_data_low*4

in i2c.py, around line 269, increasing 4 to a larger value to see if that helps.

I'll have a closer look at this issue whenever I get some spare time.

PetteriAimonen, 2017-11-27:
No problem, thanks for responding :)

Actually I found that line eventually and increasing the time fixed that part of the problem, forgot to leave a comment.

Unfortunately SMBUS also has 35 us timeout so the latencies caused by USB still caused problems. I found out Adafruit's library which queues up the full transfer sequence and it worked better for my use case.

eblot, 2017-11-27:
Ok, thanks for the follow up.

Queuing is something I'd like to add to SPI and I2C, to speed up transfer. I'd better have a look at Adafruit library to check their APIs then :P

eblot, 2018-05-04:
The timings should have been fixed with v0.28.8 (ef431724ca29438791b0aa6d9e27a808258bc3cf)

closed 2018-05-04 by eblot