eblot/pyftdi #74: Syntax Error during import

sra-kmukh, 2017-11-23:
Hello,
I am trying to use the library on Windows 10 with pyusb-1.0.0 and pyserial-3.4. I want to use it for SPI-USB communication. But I get the following error when I try to import the library:

`In [3]: from pyftdi.spi import SpiController`

>   File "C:\Program Files\Anaconda3\lib\site-packages\pyftdi-0.27.2-py2.7.egg\pyftdi\usbtools.py", line 454
    print("Available interfaces:", file=out)

`SyntaxError: invalid syntax`

Whats the issue with this?

eblot, 2017-11-23:
pyftdi-0.27.2-py2.7.egg ?

As described in [documentation](http://eblot.github.io/pyftdi/requirements.html) and  in the `setup.py` file, current releases of PyFtdi require Python3.5

For some reason, your installation is adapting to parse Python3.5+ files with an incompatible Python version (2.7)

sra-kmukh, 2017-11-24:
Thanks. 
I think that's the exact reason. Its weird but I am not able to run python 3.6 as the default version. I am getting an Unsatisfiable error. 

> UnsatisfiableError: The following specifications were found to be in conflict:
  - backports_abc -> python 2.7* -> vc 9
  - python 3.6.2*
Use "conda info <package>" to see the dependencies for each package.

Will look into it!

closed 2017-11-27 by eblot