eblot/pyftdi #75: MPSSE read error in last commit

avivbur, 2017-11-27:
Hi,
Thank you for the wonderful job you are doing.
I've been using pyftdi version 0.27.1 with the C232HM evaluation board and it has been working great.
Since the recent MPSSE interface commit (a4f66748e678daf48afbfdfbc02f2d9979a4245d), I get an ftdi mpsse error (device doesn't support MPSSE) every time I try to connect to the board. When observing the EEPROM descriptor string I see an odd seemingly random string, not matching the actual value.
For now I will keep working with v0.27.1 until fixed.

eblot, 2017-11-27:
Can you post the full traceback when this error occurs? Thanks.

avivbur, 2017-11-27:
Will try to reproduce it asap

eblot, 2019-06-04:
No feedback, closing.

closed 2019-06-04 by eblot