eblot/pyftdi #76: SPI features

avivbur, 2017-11-27:
Again,
Thank you for this useful code base.
Current SPI configuration does not allow pulling up cs line after every word is sent as is custom with many SPI slaves. Current SPI configuration also has a set time after every transaction before the cs line is pulled (regardless of protocol frequency), instead of waiting a configurable number of **SPI clock cycles** 
So really the features I suggest adding are:
**1. Allow configuring SPI to raise CS line after every word sent, and not only at the end of the transaction .
2. Wait a configurable amount of spi clock cycles before raising CS line after every transaction (default should be one)**

eblot, 2017-11-27:
I'm surprised about 1:

`/CS` line is released (high level) on each transaction (i.e. `read` or `write` or `read/write`), except if you change the default `start` and `stop` arguments.
If you want to release `/CS` on each word, call write() with a single word as the argument.

avivbur, 2017-11-27:
Yes I managed to use some workarounds of this sort, but the fact is that a read transaction uses a minimum of 2 words, and burst writes cannot be used this way. Some spi slaves use CS falling edge as a condition between words, hence making this feature important.

eblot, 2017-11-27:
Could you provide an example of what you're attempting to achieve? I think I do not get the actual issue.

eblot, 2018-01-17:
No news, closing. Feel free to re-open if you can provide some details about this issue.

closed 2018-01-17 by eblot