eblot/pyftdi #78: Full duplex mode ??

MadaniJo, 2017-12-14:
Dear Sir,
Is there any easy work around to get the FTDI ( FT232H in our case ) working in full duplex mode ?
Unfortunately we discovered a bit late that the sensor we want to use is a full duplex device and we started a board.
Seasons Greetings.
Best regards,
DM

eblot, 2017-12-18:
Full duplex mode with which mode? SPI?

A preliminary version for SPI full duplex is available in the master branch. Have you given it a try?

MadaniJo, 2017-12-19:
Bonjour Emmanuel,
OK I didn't see this. I checked out the Master and ... it worked 👍.
I made a quick check at 6M and 10MHz in mode 0. So far I am not sure if the first byte gets lost or not, but I let you know. Thanks this was a great suggestion.
Joyeux Noel.
Salutations,
Madani

eblot, 2017-12-20:
Ok, great.
Let me know if you encounter any issues, I need to update the documentation BTW.
Merci et bonnes fêtes également.

eblot, 2018-01-15:
Closing this issue as full duplex implementation is now (v0.28.2) available.

closed 2018-01-17 by eblot