eblot/pyftdi #79: Access ACBUS and use it as GPIO?

lifetyper, 2017-12-18:
As I see the GPIO library only handle 8bits of the ADBUS,could it be possible to add ACBUS support in the GPIO part?

I used to use FTDI library from adafruit but that only works with python2,and I'm sure ACBUS is accessible to software interface.

lifetyper, 2017-12-18:
I found some useful code as a work around in issue 38 by chadkennedyonline:
https://github.com/eblot/pyftdi/issues/38

but I'm still wondering is it possible to add acbus control in library?

eblot, 2017-12-18:
Yes it should be possible using the `SET_BITS_HIGH` command. It has already been implemented with SPI mode.

lifetyper, 2017-12-20:
Thanks a lot,I can configure the ACBUS using SpiController().get_gpio() now.

eblot, 2017-12-20:
Re-opening this feature request as it would be great to provide GPIO support for ACBUS with bit bang mode as well.

darlington72, 2018-02-26:
Dear all,

Finally, is it possible to control both SPI on AD0/1/2 and CS on AD4 + an extra gpio output on AC3 with an FT2232D?
In my case, the SPI perfectly works with AD0/1/2 and CS on AD4 but I can't add the extra gpio output on AC3.

Thank you!

eblot, 2018-02-27:
There was an issue with ACBUS, it should now be supported (v0.28.7), which has been tested with a FTDI2232H.

I do not have a FT2232D at hand - this is a really legacy device, I'm afraid I cannot test it by now.

eblot, 2018-02-28:
/CS are supposed (at least with PyFtdi) be in-order, starting from the first pin after the three mandatory SPI signals (AD0/AD1/AD2). If you do not use AD3 for /CS, but you use AD4 for the first /CS signal, AD3 is still reserved for another /CS and cannot be re-used for GPIOs, i.e. in your case:
  * AD0: SCLK
  * AD1: MOSI
  * AD2: MISO
  * AD3: N.C.
  * AD4: /CS
  * AD5,AD6,AD7: GPIOs
  * AC0,AC1,AC2,AC3: GPIOs

You still need to declare 2 /CS signal, and only use the second one, i.e.

     self._spi = SpiController(cs_count=2)  # AD3, AD4
     port = self._spi.get_port(1, freq=6E6, mode=3)  # port 0 is AD3, not used

then if you want to re-use the SPIO GPIO unit test, you need to perform the test on only 3 pins instead of 4, as there are only 3 AD pins available for GPIO, i.e. something roughly like this:
  
     ad_pins = ((1 << 3) - 1) << 5  # AD5-AD7 as input
     ac_pins = ((1 << 3) - 1) << 9  # AC1-AC3 as output
     io_pins = ad_pins | ac_pins

     # ...

     for ac_ in range(1 << 3):  # only 3 pins
        ac = ac_ << 1           # skip LSB (AD4/AC0)
        # ...

with:
   * AD5 connected to AC1
   * AD6 connected to AC2
   * AD7 connected to AC3
   * AC0 not connected

darlington72, 2018-02-28:
I have an equipment (fpga) behind the ft2232D, so when I deassert and reassert AC gpios, I have Something different than the example. But in my special case, I confirm that your example

eblot, 2018-03-02:
Ok, thanks for the feedback!

closed 2018-03-02 by eblot