eblot/pyftdi #81: Does pyftdi support FTDI's FIFO mode?

jamboo2017, 2018-01-12:
as title, thx

eblot, 2018-01-15:
Unfortunately, no. I never encountered this need up to now.

Could you describe a typical use case?

jamboo2017, 2018-01-17:
e.g. for time critical tasks like printing, the nozzles need to be controlled simultaneously and timely. :)

eblot, 2018-01-17:
Ok, I got it. It would be a nice addition indeed...

I wonder what would be the best (& easy) way to test/debug this. Any idea?

darlington72, 2018-03-12:
Hi,
FIFO modes includes the 245 FIFO Mode with RXF#, RD#, TXE#,WR and Data[7:0]?
We can't do that kind of access? https://embdev.net/attachment/114987/burst.PNG
What could be the performance with bit bang mode or just gpio mode?

Thank you

zapta, 2024-03-20:
I am using FT232H in serial mode at 4Mhz to transmit information from an FPGA to a PC python program and need x2 to x3 rate. The async FIFO mode seems to be a good fit is its require a simple parallel hardware interface, and since it's not supported, any suggestions on how I should go about implementing it? 

For example, is it reasonable implementing it on top of the raw API https://eblot.github.io/pyftdi/api/ftdi.html ?  Any other suggestions?