eblot/pyftdi #88: I2C read timing issue with FT2232H

alec178, 2018-02-22:
I met a pyftdi I2C problem with FT2232H.

When I use pyftdi I2cPort.read(16), SCL timing interval between each read byte is too closed (see below waveform). It makes the read data sometimes right and sometimes wrong.
Is there any way to enlarge SCL timing interval of each byte?

[image]

eblot, 2018-02-22:
This is something that needs to properly get fixed (as it depends on the I2C bus clock)

Meanwhile, try increasing the multiplier value (from 4 to 8 for example) in the following lines (i2c.py@269) 
        self._start = data_low*4 + clock_low_data_low*4
        self._stop = clock_low_data_low*4 + data_low*4 + self._idle*4

Let me now if that helps.

alec178, 2018-02-22:
OK I will try.

But.... the "time interval between each read byte" is not related to timing of SRART/STOP bits.
The read protocol is: S-SlaveAddr-R-a-"D0-a"-"D1-a"-"D2-a"-....-"D15-n"-P
And what I want is: S-SlaveAddr-R-a---"D0-a"---"D1-a"---"D2-a"---....---"D15-n"---P

Note: 
S=Start
P=Stop
a=Ack
n=Nack

eblot, 2018-02-22:
It is likely X in `self._idle * X` you want to change

eblot, 2018-02-22:
Ok I think I got your point.

Are you looking for something like the following capture?

[image]

Could you give a try to the `i2c_read_pacing` branch?

alec178, 2018-02-23:
i2c_read_pacing works, that's what i want. Thanks a lot!

One more question, what does I2cController.read() deal with "SDA signal" after sending cmd?
From the waveform, SDA will be pull-low by either FTDI or slave side.
(I assume my slave side won't touch SDA until next clock sending from host)

[image]

eblot, 2018-02-23:
Great news, I'll clean up thinks and deliver this fix to the master branch as soon as I have some spare time :-)

eblot, 2018-02-26:
I still need to clean up / deliver to master :-)

eblot, 2018-02-26:
On Fri, Feb 23, 2018 at 4:14 AM, alec178 <notifications@github.com> wrote:

> i2c_read_pacing works, that's what i want. Thanks a lot!
>
> One more question, what does I2cController.read() deal with "SDA signal"
> after sending cmd?
> From the waveform, SDA will be pull-low by either FTDI or slave side.
> (I assume my slave side won't touch SDA until next clock sending from host)
>
> [image: normal]
> <https://user-images.githubusercontent.com/32443068/36576252-7835657e-188a-11e8-8b9f-50207a06f193.jpg>
>
>
> Hi,

I do not observe this SDA low signal on my current setup:

first two signals: SCL, SDA digital
last two signals: SCL, SDA analog

I remember bumping into weird I/O levels with I2C pull up resistors that
has too strong values, but I’ve not been able to understand the observed
levels looking at the actual code…

It may be worth tweaking the pull up to see if the signal improves. This
seems to especially true when the resistor between the SDI and SDO is << to
the pull up resistors on each line.

Cheers,
Manu.

eblot, 2018-02-26:
i2c_read_pacing delivered to master as v0.28.6

eblot, 2018-02-26:
[image]

closed 2018-02-26 by eblot

alec178, 2018-02-26:
Hi Emmanuel,

Thanks for your prompt reply.

The I2C slave device I am using is a customized FPGA.
There are two problems I suffered:

*(1) In normal case:*
SDA was pulled low, 83us right after the 9th clock in the read cmd.
In general design, SDA shall be free (goes back to HIGH) after the 9th
clock.
It's hard to debug which side pullls low the SDA. So I would like to know
your comment first.

​
​

*(2) In abnormal case:*
I did a simple experiment.
Host sends a read cmd to slave (FPGA), and force slave not send any data to
host except the ack.
Then I will see the strange waveform as below:

​

Alec