eblot/pyftdi #89: windows does not honor serial number in url

rdpoor, 2018-02-22:
Attached is a minimal python script that locates all the FTDI FT232R devices on the system and then tries to open one device, with and without a serial number in the URL.  

In summary, on a Windows system with two FT232R devices plugged in:

* With a serial number in the URL, create_from_url() erroneously raises the error `No USB device with S/N A932QG17A`
* Without a serial number in the URL, create_from_url() correctly raises the error `2 USB devices match URL`

Here is a printout of the the results of the attached python script:

```
E:\>py -i ftdi-serial-number-test.py
Found the following ports with FT232R parts:
{
  "device": "COM9",
  "name": null,
  "description": "USB Serial Port (COM9)",
  "hwid": "USB VID:PID=0403:6001 SER=A932QG17A",
  "vid": 1027,
  "pid": 24577,
  "serial_number": "A932QG17A",
  "location": null,
  "manufacturer": "FTDI",
  "product": null,
  "interface": null
}
{
  "device": "COM11",
  "name": null,
  "description": "USB Serial Port (COM11)",
  "hwid": "USB VID:PID=0403:6001 SER=FT3BNM1OA",
  "vid": 1027,
  "pid": 24577,
  "serial_number": "FT3BNM1OA",
  "location": null,
  "manufacturer": "FTDI",
  "product": null,
  "interface": null
}
opening device with url =  ftdi://0x403:0x6001:A932QG17A/1
Error: No USB device with S/N A932QG17A
opening device with url =  ftdi://0x403:0x6001/1
Error: 2 USB devices match URL
```
(Note: the .py script has been renamed with a .txt script so github will accept it as an attachment.)

[ftdi-serial-number-test.txt](https://github.com/eblot/pyftdi/files/1749577/ftdi-serial-number-test.txt)

rdpoor, 2018-02-22:
I have a little time and will dig into the code to see if I can identify the problem...

rdpoor, 2018-02-22:
I put a breakpoint inside usbtools._find_devices():251 and looked at the raw device list.  My impression is that `dev.iSerialNumber` isn't getting set to what usbtools expects.  In my test, it's set to `3` when the real serial number of the device is `A932QG17A`.

`iSerialNumber` is generated somewhere in the backend.  But that's as far as I can get today.

```
> ...\site-packages\pyftdi\usbtools.py(251)_find_devices()
-> if platform == 'win32':
(Pdb) p devs
{<DEVICE ID 0403:6001 on Bus 000 Address 003>, <DEVICE ID 0403:6001 on Bus 000 A
ddress 002>}
(Pdb) p list(devs)[0].__dict__
{
  '_finalize_called': False,
  '_finalizer': <finalize object at 0x439e80; for 'Device' at 0x1d08190>,
  '_ctx': <usb.core._ResourceManager object at 0x01D080D0>,
  '_Device__default_timeout': 1000,
  '_serial_number': None,
  '_product': None,
  '_manufacturer': None,
  '_langids': None,
  'bLength': 18,
  'bDescriptorType': 1,
  'bcdUSB': 512,
  'bDeviceClass': 0,
  'bDeviceSubClass': 0,
  'bDeviceProtocol': 0,
  'bMaxPacketSize0': 8,
  'idVendor': 1027,
  'idProduct': 24577,
  'bcdDevice': 1536,
  'iManufacturer': 1,
  'iProduct': 2,
  'iSerialNumber': 3,
  'bNumConfigurations': 1,
  'address':3,
  'bus': 0,
  'port_number': None,
  'port_numbers': None,
  'speed': None
}
```

eblot, 2018-02-26:
Windows (USB support and many other topics) are still a mystery for me, so I would really  welcome any patch, but I do not think I'll have time to work with this OS.

rdpoor, 2018-02-26:
Emmanuel: Heh!  Windows is just as mysterious to me :) but I have a client who needs this, so I'm motivated to find a solution for all.  Schedule willing, I'll be digging into it again later today.  `pdb` is my friend...

rdpoor, 2018-02-28:
Okay.  I have *observed* the bug, but I haven't fully identified it.  The serial number of the device is `FT3BNM1OA`, and that's what I'm passing in on the URL:

    ftdi://0x403:0x6001:FT3BNM1OA/1

But in pyftdi/usbtools.py(345) around this code:

        if sernum:
            if sernum not in [dev[2] for dev in devices]:
                raise UsbToolsError("No USB device with S/N %s" % sernum)

... sernum is `FT3BNM1OA` and dev is `(1027, 24577, 'FT3BNM1O', 1, 'USB-RS485 Cable')` under Windows.  (Note the missing 'A' in the latter's serial number.)  I've traced down into ftdi/util.py, but it's not clear how or why the serial number is getting truncated.  Limited to 8 chars on Windows?

Still digging...

rdpoor, 2018-02-28:
Ah.  There appears to be a disagreement originating in comports() under windows:

    serial.tools.list_ports.comports()[0].serial_number => FT3BNM1OA on Windows, FT3BNM1O on Mac
    UsbTools.get_string(dev, dev.iSerialNumber)` => FT3BNM1O on both platforms

rdpoor, 2018-02-28:
Okay -- after digging deeper, this turns out to be an issue with `pyserial's comports()` -- its the one returning an incorrect serial number.

See https://github.com/pyserial/pyserial/issues/283 for a description of the root cause.

Closing this one.

closed 2018-02-28 by rdpoor