eblot/pyftdi #90: Maximum data length of one I2C write?

alec178, 2018-02-26:
I send an I2C write with data length > 255 bytes, by function `I2cController.write(self, address, out, relax=True)`. And then I met an OverflowError as below.
Is there anyway to break the max length limitation of the byte array?

```
    self.port.write(tx_buf)
  File "C:\Python36\lib\site-packages\pyftdi\i2c.py", line 120, in write
    relax=relax)
  File "C:\Python36\lib\site-packages\pyftdi\i2c.py", line 457, in write
    self._do_write(out)
  File "C:\Python36\lib\site-packages\pyftdi\i2c.py", line 680, in _do_write
    out = array('B', out)
OverflowError: unsigned byte integer is greater than maximum
```

eblot, 2018-02-26:
The error looks like an attempt to write a "byte" value greater than 255, not a buffer overflow. Can you print out the value(s) you are trying to send?

alec178, 2018-02-26:
Yes, you are right. It's my coding fault.
Thank you.

closed 2018-02-26 by alec178