eblot/pyftdi #91: Host send a read cmd to slave, but none data is ready to send from slave

alec178, 2018-02-26:
I am using a customized I2C slave device. And I get a strange waveform with the test scenario:
1. Host (FT2232H) sends a read cmd to slave, `I2cController.read(2)`
2. In case slave has "none" data to send except for the ack

Then I will see the waveform as below.
My question is, what will pyftdi deal with SDA for such condition?

[image]

eblot, 2018-02-26:
I'm not sure how to interpret the above screenshot, as the SCL signal seems weird.
1. First L-H-L pulse is definitely a glitch, it is very hard to avoid them with the way FTDI works, but it should not be an issue
2. Second L-H-L pattern looks like the ACK from read, which the master (here: FTDI) should emit
3. NACK (at the end of the last received byte) is not seen, as expected
4. the last L-H-L pattern looks like a STOP condition, but as SCL does not seem to return to full H level, this is weird.

eblot, 2018-05-04:
I do not know how to reproduce this issue.

I'm wondering: does the slave use clock stretching (i.e. maintain SCL low)? This feature is not implemented in PyFTDI for now...

closed 2018-05-05 by alec178