eblot/pyftdi #94: usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

erkandiken, 2018-03-27:
Hello,

I have been facing the following error::
....
  File "//anaconda/envs/adaptive_module_conda_env/lib/python3.5/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

while executing the following command:
-> i2c.configure(url) 

When I list the connected device:
i2c.configure('ftdi:///?')

This is what I get as url:  ftdi://ftdi:232h:FT2H2BPY/1   (UM232H) 

So I use this url as parameter for the configure function.

I am on macOS Sierra (10.12.6).

I am stuck on this, any idea to solve this issue ?

eblot, 2018-03-27:
Have you read/applied
http://eblot.github.io/pyftdi/troubleshooting.html#error-access-denied-insufficient-permissions ?

eblot, 2018-04-25:
No reply, closing.

closed 2018-04-25 by eblot