eblot/pyftdi #97: Nack in i2c read when transfer is split in multiple chunks

Rico-van-Dongen, 2018-04-16:
When I try to read a large number of bytes such that the I2C read needs to split into multiple chunks a NACK is send in-between the chunks. Due to the NACK the slave releases the bus and only 0xFF is read in the remaining chunks. 
When I change the behavior of the _do_read function in I2C.py such that the read_last also sends an ACK I receive all the data. However, then also the very end of the transfer an ACK is send while this should be an NACK to end te transfer. 
Could you please verify the splitting of the read-transfer into chunks. I guess chunks 0 to N-1 should Not include the read_last but chunk N should include it.

Rico-van-Dongen, 2018-04-25:
I made the following changes to make it work for our application. 
[i2c.txt](https://github.com/eblot/pyftdi/files/1947483/i2c.txt)

eblot, 2018-05-02:
Hi Rico,

I'm not sure to understand your changes (BTW, a pull request or a diff if you do not want to use Git is far better to see the modification you propose):

    --- i2c.py  2018-05-02 18:10:15.000000000 +0200
    +++ i2c.txt 2018-05-02 17:18:56.000000000 +0200
    @@ -659,7 +659,11 @@
                 if last_count != block_count:
                     cmd = array('B')
                     cmd.extend(read_not_last * (block_count-1))
    -                cmd.extend(read_last)
    +           if count <= 0:
    +              cmd.extend(read_last)
    +           else:
    +              # only append the read_last (send nack) on last byte of last chunk
    +              cmd.extend(read_last)
                     last_cmd = cmd
                 else:
                     cmd = last_cmd

It seems to be a no-change, as whatever `count` value, the `if` body is the same.

Rico-van-Dongen, 2018-05-02:
Hi Emmanuel

I think you got the essence of the change quite right. Code-wise the long-read branch looks fine to me. It is actually a bit more readable than my change since you combined the read-last and the  self._immediate in the same if. 
Actually looking at the diff you posted I think I made a copy-paste error in my change proposal. There should be a read_NOT_last in the else instead of another read_last.  
I don’t have the opportunity to test the branch on hardware any time soon, however I’m fairly confident that your change will fix te problem. Let see if it makes any differences for #101, if so you can consider this issue fixed as well.

eblot, 2018-05-04:
Ok Rico, thanks for the feedback. I'm closing this one as a duplicate (even if it is the first report, as #101 as a chronogram that helps describing the issue. Thanks.

closed 2018-05-04 by eblot