eblot/pyftdi #100: How to enlarge the timing interval b/w last Ack and 1st data in?

alec178, 2018-05-02:
Hi Emmanuel,

I'm using pyftdi-0.28.8.
I would like to enlarge the timing interval b/w last Ack and 1st data in, within an I2C READ. (see below diagram, for example, make interval time 216us to 1ms)
How shall I configure `pyftdi`?

[image]

Alec

eblot, 2018-05-04:
Well, it is a bit out-of-scope with the initial design, but I think I could add an API that would allow this kind of use case among others.

eblot, 2018-05-04:
I've added a new boolean parameter to I2cPort method: start(=True)
If it is unset, PyFtdi will not emit a prolog (start + address) sequence.

Therefore you could write:

    from time import sleep

    # emit a START sequence is read address, but read no data and keep the bus busy
    port.read(0, relax=False)
    # wait for ~1ms (0.8ms + imprecise USB stack delay) 
    sleep(0.0008) 
    # read 4 bytes, without emitting the start sequence, and release the bus
    port.read(4, start=False)

Would it be ok?

See `i2c_opt_start` branch.

alec178, 2018-05-05:
Sorry, I can't find any branch, which is named `i2c_opt_start `.

[image]

eblot, 2018-05-05:
Sorry, I forgot to push it. Now it is.

eblot, 2018-05-15:
`i2c_opt_start` has been merged into `master` branch as v0.29.0

closed 2018-05-15 by eblot

alec178, 2018-05-16:
It works, thanks!

eblot, 2018-05-16:
Great, thanks for the feedback