eblot/pyftdi #101: An unexpected NACK is sent by pyftdi read(), once total read bytes > 96

alec178, 2018-05-02:
Hi Emmanuel,

I'm using pyftdi-0.28.8.
I would like to continuously read more than 96 bytes from an I2C slave.
But starting from 97th byte, the read data is all 0xFF.
Look into the waveform, after the 96th byte data is read back, pyftdi sends a NACK to slave.
Could you give me hints by using pyftdi I2C read() API.

My pseudo code is listed below.

```
    def r32_inc(self):
        buf = [0x00, 0x00, 0x00, 0x00] # write address
        self.port.write(buf)

        val = self.port.read(100) # expect to read 100 bytes
        return val
```

[image]

eblot, 2018-05-02:
It might be the same issue as with #97.

I never tested the implementation with large transfer, it is time to create such as test environment...

Rico-van-Dongen, 2018-05-02:
Hi alec, 
It indeed sounds like the issue #97 that I reported. You could try to modify the code according to the change I proposed and see if that works for you as well.

alec178, 2018-05-05:
Hi Emmanuel,

It works with` i2c_long_read` branch.
Thanks! 👍