eblot/pyftdi #102: Simultaneous SPI and GPIO access

koffes, 2018-05-25:
I have a question which may turn out to be a feature request.

We have a SPI peripheral connected to ADBUS0=CLK, ADBUS1=MOSI, ADBUS2=MISO, ADBUS7=CS.
In order to communicate with this device: 
`spi = SpiController(cs_count=5)
 spi.configure('ftdi://ftdi:2232h/1')
slave = spi.get_port(cs=4, freq=1E6, mode=0)`

This seems OK, but now I need to set a GPIO on ADBUS6 duiring the SPI transfer. Let's call it an additional CS pin. How can this be done? I have attempted with `gpio.set_direction(0x40, 0x40)` but then an error occurs because one is messing with SPI reserved pins. No luck with `spi._ftdi.write_data(Array('B', [Ftdi.SET_BITS_LOW, 0x00, 0x40]))` either.
Is is possible to either:
1. Set cs_count=1 and just use ADBUS7 and leave the other pins between vacant?
2. Use two CS pins at the same time?
3. Get access to pins as GPIO's within the SPI CS range?

Toggling pins on ACBUSx works fine.

eblot, 2018-05-25:
PyFtdi is designed to use /CS signals that are physically located to SCLK/MOSI/MISO lines. Every line between MISO and the last /CS are reserved for SPI by (pyftdi) design I'm afraid.

You would either have to hack the code, or rework the HW. I do not plan to add this feature, but patches are welcome :-). I'm not sure it could be implemented w/o altering the existing API, which would be an issue.

koffes, 2018-05-25:
Thank you for the quick response. I will have to get around this by telling the SPI to use ADBUS6 as CS and then manually setting the real CS on ADBUS7 low during SPI transfer.
There is an example here http://eblot.github.io/pyftdi/api/spi.html how to use GPIO and SPI at the same time.
However, the bit mask is only 8 bits wide. How do you switch between accessing GPIOs on ADBUS and the ACBUS? I need to control ADBUS7, ACBUS0 and ACBUS7.

eblot, 2018-05-25:
The FTDI MPSSE command set use different commands for AD and AC:
``Ftdi.SET_BITS_LOW`` for AD, ``Ftdi.SET_BITS_HIGH`` for AC.

koffes, 2018-06-04:
I managed to get around the issue quite easily by overwriting the class variable `_spi_mask`:
`self._ctrl = SpiController(silent_clock=False, cs_count=5)`
`self._ctrl._spi_mask = self._ctrl.SPI_BITS | (1<<(cs+NUM_SPI_PINS))`
If I find the time I will submit a more elegant PR.

closed 2018-06-04 by koffes