eblot/pyftdi #104: Accessing to interface A and B of FT2232H

dbellix, 2018-07-17:
Hello,
Thanks for you job in providing library for FTDI devices.
I have a FT2232H, that I am trying to use it with python. The main problem is accessing to the two MPSSE engines within the chip. Even following the examples for the ft2232h:

ctrl.configure('ftdi://ftdi:2232h/1')

and changing it to

ctrl.configure('ftdi://ftdi:2232h/0')

if I change it to ctrl.configure('ftdi://ftdi:2232h/2'), it says "No such FTDI port 2".
It does not make difference. I am getting always the output from the same port (BD). It seems that the 'interface' field on the url does not work properly. This has been done with the latest version of the pyftdi package (0.29.0 using libusb1) on Win10 64-bit. I tried also to switch to Python 2.7 with the last package supported  (0.13.4), and changing the value of 'interface' as described above, but it seems to not be effective.

Do you have any idea to workout this problem? Thanks in advance

Best,
Davide

eblot, 2018-07-21:
Hi, 

`/0` is a legacy syntax which should not be used.
`/1` and `/2` is the proper syntax to use.

Did you get the chance to try this on a non-Windows host? Windows is just a mess with (lib)usb, so depending on how you have defined the USB port, you may get different results (this is why Windows is not officially supported: I do not want to lose time with this OS).
Could it be somehow related to #63?

eblot, 2019-06-04:
No feedback, closing.

closed 2019-06-04 by eblot