eblot/pyftdi #109: SPI Mask incorrectly set

JarrettR, 2018-11-09:
My code (mostly from example):

```
        # Instanciate a SPI controller
        self.spi = SpiController()

        # Configure the first interface (IF/1) of the first FTDI device as a
        # SPI master
        self.spi.configure('ftdi://::/1') 

        # Get a SPI port to a SPI slave w/ /CS on A*BUS3 and SPI mode 0 @ 12MHz
        self.slave = self.spi.get_port(cs=0, freq=1E6, mode=0)

        # Get GPIO port to manage extra pins, use A*BUS4 as GPO, A*BUS4 as GPI
        self.gpio = self.spi.get_gpio()
        print("-- ")
        print(self.spi._spi_mask)
        print(self.spi._cs_bits)
        print(self.spi.CS_BIT)
        print(self.spi.SPI_BITS)
        print(self.spi._cs_count)
        print("-- ")
        self.gpio.set_direction(0x10, 0x10)
```

Output:

```
--
127
120
8
7
4
--

...

 File "/home/j/spi.py", line 43, in __init__
    self.gpio.set_direction(0x10, 0x10)
  File "/usr/local/lib/python3.6/dist-packages/pyftdi-0.29.2-py3.6.egg/pyftdi/spi.py", line 230, in set_direction
    self._controller.set_gpio_direction(pins, direction)
  File "/usr/local/lib/python3.6/dist-packages/pyftdi-0.29.2-py3.6.egg/pyftdi/spi.py", line 433, in set_gpio_direction
    raise SpiIOError('Cannot access SPI pins as GPIO')
pyftdi.spi.SpiIOError: Cannot access SPI pins as GPIO
```

The issue seems to be around here:

https://github.com/eblot/pyftdi/blob/93eb8b049287acc79e441fe2c33f9724450502b4/pyftdi/spi.py#L298

See above, where `SpiController()_cs_bits` is set to 120 right out of the gate.

Jallen1990, 2019-06-03:
> 
> 
> So, on further inspection, the masking seems right, the issue is that cs_count is [set to 4](https://github.com/eblot/pyftdi/blob/93eb8b049287acc79e441fe2c33f9724450502b4/pyftdi/spi.py#L249) by default.
> 
> I've updated the documentation in PR #110 to show correct handling of this.

I also meet this issue, and solved by your recommend：
SpiController(cs_count =1)
it works well
thanks so much!

eblot, 2019-06-03:
Sorry for some reasons I missed the notifications for this project.

I think it would be safer to use `cs_count` default value as 1, and document this setting. There is no real reason for using another default CS count value.

eblot, 2019-06-04:
I think this change will be deferred of the next "major" version, as it may break existing installation that rely on the default value. As I plan to add some other changes, such as returning bytes instead of Array('B'), I'd like to limit the number of potential API breaks.

JarrettR, 2019-06-05:
That's fair, easy enough to leave this issue open for now or change the title to reflect that the default value is to be changed.

The PR with the documentation change is non-breaking and a good to stopgap to merge immediately though!

sc-splitter, 2019-09-25:
I'm having a similar issue as well. I'm trying to use both the SPI and GPIO at once, and no matter what I try to set the GPIO pin directions to, the mask remains at 0x0000. See below:

``` python
from pyftdi.spi import SpiController

if __name__ == '__main__':
    # Get a controller instance
    control = SpiController(cs_count=2)
    
    # Connect to the FTDI chip
    control.configure('ftdi://ftdi:232h/1')
    
    # Get a SPI handle
    spi = control.get_port(cs=0, freq=10e6)
    
    # Print the SPI mask
    print(f'---\nSpi mask: 0x{control._spi_mask:04x}')
    
    # Get a GPIO handle and set the pins
    gpio = control.get_gpio()
    gpio.set_direction(0xFFE0, True)
    
    # Print the GPIO mask and direction
    print(f'GPIO mask: 0x{control._gpio_mask:04x}')
    print(f'GPIO direction: 0x{control._gpio_dir:04x}\n---')
    
    # Write all the GPIO pins high
    gpio.write(0xFFE0)
```

Output:
```
---
Spi mask: 0x001f
GPIO mask: 0xffe0
GPIO direction: 0x0000
---
Traceback (most recent call last):
  File "ftdi_test.py", line 25, in <module>
    gpio.write(0xFFE0)
  File "C:\Python\envs\py3\lib\site-packages\pyftdi\spi.py", line 234, in write
    return self._controller.write_gpio(value)
  File "C:\Python\envs\py3\lib\site-packages\pyftdi\spi.py", line 443, in write_gpio
    (self._gpio_dir, value))
pyftdi.spi.SpiIOError: No such GPO pins: 0000/ffe0
```

eblot, 2019-09-25:
@sc-splitter: You are not calling `set_direction` the proper way:
     gpio.set_direction(0xFFE0, True)

Check out the [API](https://eblot.github.io/pyftdi/api/spi.html#pyftdi.spi.SpiGpioPort.set_direction):
  * `direction` should be specified as a bitfield, not a `bool`. You are declaring that SCLK is the only output pin.

closed 2019-11-05 by eblot