eblot/pyftdi #113: Typo in gpio.py

snizzard, 2018-12-04:
There seems to be a typo in `gpio.py` in the `set_direction` definition.

Here is the whole function:

`    def set_direction(self, pins, direction):
        """Update the GPIO pin direction.

           :param int pins: which GPIO pins should be reconfigured
           :param int direction: a bitfield of GPIO pins. Each bit represent a
                GPIO pin, where a high level sets the pin as output and a low
                level sets the pin as input/high-Z.
        """
        if direction > self.MASK:
            raise GpioException("Invalid direction mask")
        self._direction &= ~pins
        self._direction |= (pins & direction)
        self._ftdi.set_bitmode(self.direction, Ftdi.BITMODE_BITBANG)`

In that last part, the function changes the global variable `self._direction` but then passes the local variable `direction` to the function.  This seems odd...

Shouldn't it be passing the global `self._direction` with an underscore instead of `self.direction` ?

eblot, 2018-12-06:
self.direction is a property that returns self._direction (see @ line ~55), so reading one or another value does the same, but I agree it would be cleaner to call the instance variable rather than the property, I'll change this.

Thanks for the report.

eblot, 2019-06-04:
Done in 0a451f63

closed 2019-06-04 by eblot