eblot/pyftdi #114: usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

blancyang, 2018-12-13:
When I  open the device but I got the following error

Environment:
OS: macOS 10.13.6
libusb 1.0.22
FT232H

```
Traceback (most recent call last):
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1155, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1450, in _write
    return self.usb_dev.write(self.in_ep, data, self.usb_write_timeout)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 940, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "i2c_tool.py", line 10, in <module>
    i2c.configure('ftdi://ftdi:0x6014/1')
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/i2c.py", line 359, in configure
    frequency=frequency, **kwargs)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 475, in open_mpsse_from_url
    direction, initial, frequency, latency, debug)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 537, in open_mpsse
    frequency = self._set_frequency(frequency)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1586, in _set_frequency
    self.write_data(cmd)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1161, in write_data
    raise FtdiError('UsbError: %s' % str(ex))
pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)

```

System returns the following message when trying to unload the driver 
```
bash-3.2$ kextunload -bundle com.apple.driver.AppleUSBFTDI
(kernel) Kext com.apple.driver.AppleUSBFTDI not found for unload request.
Failed to unload com.apple.driver.AppleUSBFTDI - (libkern/kext) not found.
```

eblot, 2018-12-13:
Weird, I've been using FT232H on macOS 10.13 on a daily basis with no issue.
I'm now runing 10.14.2 so I cannot reproduce the environment - where unloading the kernel extension is no loger required to run PyFTDI.

Did you run the kextunload as root? Have you checked that no other software is using the FT232H when you execute kextunload?

blancyang, 2018-12-13:
I ran kextunload as root. I decide to move to 10.14 see if I still get the same problem again.

blancyang, 2018-12-13:
Still get the same error after updating to 10.14.2

Before unload kext
```
Traceback (most recent call last):
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1155, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1450, in _write
    return self.usb_dev.write(self.in_ep, data, self.usb_write_timeout)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 940, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "i2c_tool.py", line 10, in <module>
    i2c.configure('ftdi://ftdi:0x6014/1')
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/i2c.py", line 359, in configure
    frequency=frequency, **kwargs)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 475, in open_mpsse_from_url
    direction, initial, frequency, latency, debug)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 537, in open_mpsse
    frequency = self._set_frequency(frequency)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1586, in _set_frequency
    self.write_data(cmd)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1161, in write_data
    raise FtdiError('UsbError: %s' % str(ex))
pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)
```

after unload kext

```
Traceback (most recent call last):
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1155, in write_data
    length = self._write(data[offset:offset+write_size])
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1450, in _write
    return self.usb_dev.write(self.in_ep, data, self.usb_write_timeout)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 940, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 13] Access denied (insufficient permissions)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "i2c_tool.py", line 10, in <module>
    i2c.configure('ftdi://ftdi:0x6014/1')
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/i2c.py", line 359, in configure
    frequency=frequency, **kwargs)
[... cut ...]
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 537, in open_mpsse
    frequency = self._set_frequency(frequency)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1586, in _set_frequency
    self.write_data(cmd)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/ftdi.py", line 1161, in write_data
    raise FtdiError('UsbError: %s' % str(ex))
pyftdi.ftdi.FtdiError: UsbError: [Errno 13] Access denied (insufficient permissions)

```

blancyang, 2018-12-13:
I install libusb by "brew install libusb"

eblot, 2018-12-13:
Ok, as I always do.
I really have no idea what's going on with your setup, I never encountered this issue while PyFTDI is used with dozens of Mac in my office.
Check your system log file to see if you can spot something unexpected, and it may be worth to define `LIBUSB_DEBUG` (http://libusb.sourceforge.net/api-1.0/#msglog) to see if some message from the lower stacks could give a clue.

blancyang, 2018-12-13:
Issue is fixed after I try a VM with clean OSX installation.

closed 2018-12-20 by blancyang