eblot/pyftdi #116: NACK when sending slave.read_from(0x00, 1)

blancyang, 2018-12-20:
I use the following code to test BMI160 chip with FT232H. Read the chip id.

```
from pyftdi.i2c import I2cController

# Instanciate an I2C controller
i2c = I2cController()

# Configure the first interface (IF/1) of the FTDI device as an I2C master
i2c.configure('ftdi://ftdi:0x6014/1')

# Get a port to an I2C slave device
slave = i2c.get_port(0x68)
```

However, I got the following error

```
NACK
Retry exchange
NACK
Retry exchange
NACK
Traceback (most recent call last):
  File "i2c_tool.py", line 24, in <module>
    slave.read_from(0x00, 1)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/i2c.py", line 147, in read_from
    out=self._make_buffer(regaddr), readlen=readlen, relax=relax)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/i2c.py", line 545, in exchange
    self._do_prolog(i2caddress)
  File "/Users/pyang/.local/share/virtualenvs/host_data_capture-3sxS33C5/lib/python3.7/site-packages/pyftdi/i2c.py", line 677, in _do_prolog
    raise I2cNackError('NACK from slave')
pyftdi.i2c.I2cNackError: NACK from slave
```
[image]

From logic analyzer, I saw the device response ACK. Looks like FT232H doesn't receive it.

blancyang, 2018-12-20:
BDBUS0 - BMI SCL 
BDBUS1 - BMI SDA
SCL and SDA have 10k pull up resistor. 

From your document, should I connect BDBUS1 & BDBUS2 together?

blancyang, 2018-12-20:
After connect BDBUS1 & BDBUS2 together. The problem is solved.

closed 2018-12-20 by blancyang

eblot, 2018-12-20:
FTDI pins in MPSSE mode cannot be (re)configured: the signal direction is hardcoded, so yes, you need to connect AD1 as SDA output, and AD2 as SDA input.