eblot/pyftdi #117: long delay between bytes during i2c write

Fatflo, 2019-01-17:
Hello,
I am using the latest master and a UM232 board, I observe long delays between written bytes of about ~0.5ms with a 400Khz clock.

Is it possible to decrease the delay between bytes?
[image]

eblot, 2019-01-17:
How do you write those bytes? One after another?

Fatflo, 2019-01-17:
I am writing using the following sequence, where frame is a 4bytes 'bytes' array:

slave = i2c.get_port(0x20)
slave.write(frame)

p.s running on macOS10.14

eblot, 2019-01-17:
A round trip over USB a required to send bits to the I2C master, then to read the acknowledgement (ACK/NACK) bit on each byte.

An optimisation could be implemented sending several bytes in a row, but that would imply that bytes from a frame to be sent to the slave even the slave does not acknowledge/reject one. This does not seem desirable in most use cases.

AFAICT, there is no FTDI command to stop sending a frame on first NACK, which means that the host needs to take care about every ACK bit. The delay you are observing is mostly due to the round trip on the USB bus (and some delay due to the USB stack and Python). I'm not sure it can be improved.