eblot/pyftdi #119: UsbToolsError should be in __all__

mcondarelli, 2019-01-25:
Subject says it all.

`usbtools.UsbToolsError` is raised, if needed, but trying to catch it with:

```
    from pyftdi.usbtools import UsbToolsError
    try:
        ...
    except UsbToolsError:
        pass
```

Results in a warning: `'UsbToolsError' is not declared in __all__`.

eblot, 2019-01-28:
I'm not sure how you got this warning. Are you using a linter, or some fancy editor such as PyCharm?

`__all__` is for managing `*` import, ''i.e.''

    from pyftdi.usbtools import *

Although it would make no harm to import the exception as well, I do no see why a warning would be emitted with the current definition.

mcondarelli, 2019-01-28:
I am using PyCharm indeed.

Further testing revealed I might receive two different errors (`UsbToolsError` or `USBError`) in essentially the same place, when a USB disconnect occurs.

This seems to be due to fact Your code traps (and re-rises as `UsbToolsError`) `USBError` in a few places, while in others it's not trapped, so it is exposed to my code. My hardware, unfortunately, is very prone to USB disconnections, so I'm in the best position for testing ;)

(Problem is I'm driving a hardware nail-bed to program a SPI Flash at end of a small production line, so there's a lot of noise/vibrations adversely impacting on FT4232 connection)

I also had to use `UsbTools.flush_cache()` to recover from USB disconnect. Is that enough to fully reset `ftdi`? My program is in a loop, but each cycle should completely re-initialize `ftdi` after hardware connections are stable.

Thanks

eblot, 2019-01-28:
Ok, got it. I think I'll remove `__all__` definition, which is useless anyway.

About the HW issue:

  - can you open another issue with traces where UsbError is raised and not caught? It may be on purpose or not, so it should be fixed and/or documented. It is unexpected to get USBErrors (i.e. comm errors) if the trouble comes from the communication between the SPI slave (data flash) and the SPI master (FTDI) in UsbTools, I'd like to address this.

Unfortunately, I never found a 100% successful method to reset an FTDI device. Even sending USB reset commands and buffer flush request sometimes fail to reset an FTDI master. But again, it is weird that a SPI bus issue translate into USB errors.

The cache flush is to address rediscovery of FTDI devices on a bus. If you need to call it, it seems to indicate the USB bus itself is in trouble, or you somehow shut down the FTDI power supply. Anyway, any traces would help to understand what's going on.

I'm leaving this ticket open to address the original issue with `__all__`.

mcondarelli, 2019-01-28:
Thanks for the fast answer.

Sorry I didn't make myself clear.

Problems occur at USB connection level.

Underlying (hardware) problem is MiniUSB connector on FT4232 module is not vibration resistant; manually wiggling the cable or the mechanical vibrations induced by clamp fastening board to nail-bed is enough to cause USB disconnection.
I can ascertain this as `lsusb` reports a different (increasing) Device number.

I currently have a "complex" initialization structure which opens ftdi and then tries to read a few parts of the flash in order to understand if and how it was initialized.

Apparently (I cannot test now because I have no access to hardware) if disconnect happens *before* re-opening spi: i.e.:

```python
            self._spi = pyftdi_spi.SpiController()
            self._spi.configure(self.url, cs_count=1)
            self._chn = self.spi.get_port(0)
```

I get a `UsbToolsError`, on the contrary I get the `USBError` if it happens while reading:

```python
    def read_block(self, addr, count):
        cmd = pack('BBBB', self.READ, (addr >> 16) & 0xff, (addr >> 8) & 0xff, addr & 0xff)
        return self.chn.exchange(cmd, count)

    def read(self, addr, count):
        ret = bytearray()
        while count > 0:
            u = count if count < self.channel.spi.PAYLOAD_MAX_LENGTH else self.channel.spi.PAYLOAD_MAX_LENGTH
            t = self.read_block(addr, u)
            ret += t
            addr += u
            count -= u
        return ret

    self.param = bytes(self.read(0x40000, 512))
```

Note: I have no problem in sending the whole program, if deemed useful.

Right now I somewhat trapped the errors by closing the SPI interface at end of programming (`self._spi.terminate()`), resetting USB (`UsbTools.flush_cache()`) when all boards (4) are programmed and then re-opening everything on next cycle; If you need some debugging on this I can revert to previous behavior and send error traces (probably not before next weekend, though).

eblot, 2019-01-28:
Ok, I get it.

Maybe I should rework the exception hierarchy so it is possible to catch all communication errors with a single exception.

thumbcore, 2019-01-28:
@mcondarelli I know it seems pretty clear that your USB disconnect error is mechanical. But on that topic, note that I've seen this happen on Windows 10 (not sure about 7) when the USB device was in an open state for a long period of time (hours/overnight). Turned out to be power suspend was enabled. https://www.hamoperator.com/Fusion/FusionFiles/K9EQ-Fusion-PDF-0023.pdf

If you haven't already, make sure that USB power suspend is disabled. Cheers!

mcondarelli, 2019-01-28:
Thanks @thumbcore, but I'm under Linux (Mint 19, if it matters) and my disconnects happen in matter of minutes, not hours/days.
I'm pretty sure problem is mechanical as I can easily trigger it by moving the USB cable near FH4232 connector.
My real question, here, is how to recover from disconnects without restarting the whole Python VM (of course we are also trying to limit disconnects, currently by gluing fast the "trouble-making" connection; but that is another matter entirely).
Thanks.

eblot, 2019-01-29:
As I said in one the early comments, the latter comment are not related with the initial subject of this ticket, so please open a new issue with this recovery-on-disconnection issue so we can track each issue with a dedicated ticket. I guess it would be nice to improve disconnection management.

I've removed all `__all__` statements in v0.29.3 so the initial issue should be fixed.

mcondarelli, 2019-01-29:
Done, it's issue #120.

I still get the error for:
```python
from usb import USBError
```
... but that's not on Your code, so I'll close this (I opened  pyusb/pyusb#228).

THANKS.

closed 2019-01-29 by mcondarelli

eblot, 2019-01-29:
I'm not sure why I initially added these `__all__` definitions.

I think it is better to import what is actually required, *i.e.*

    from pyftdi.usbtools import UsbTools, UsbToolsError

in this example.

Relying on `*` is a recipe for errors. This syntax is definitely not future proof, and it should be avoided IMHO.

mcondarelli, 2019-01-29:
Agreed.

In my (limited) understanding `__all__` is a way to mark elements that are part of the "public interface" of a complex module; something like omitting public/private in java declarations.

This should work similar to declaring a function/method/variable with a leading underscore, but with a module scope (`_foo` is "legal" to reference only within the single file, so it would elicit a warning if used from two different files even if in the same module).

IMHO using it is a nice feature... if it really includes all elements part of the public interface.

eblot, 2019-01-29:
There are several issues with '*': if a package is updated with a new item, the client ends up importing something it has not been designed for, and the local namespace may get polluted with this new item - with unexpected results.

One other issue is that something may be "public" for a sibling (for another module from the same package), but "private" for an external client perspective. And so on...