eblot/pyftdi #121: OSError: exception: access violation

Stimaleger, 2019-02-22:
Hi, I'm running the following script under windows 10 with python 3.7.1: 

``` python
from pyftdi.i2c import I2cController

i2c = I2cController()
i2c.configure(url='ftdi://ftdi:232h/1')
i2c.terminate()
```

And got this output :

``` console
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x000001B5DC51BAE8>
Traceback (most recent call last):
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x000001B5DC51BAE8>
Traceback (most recent call last):
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x000001B5DC51BAE8>
Traceback (most recent call last):
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
Exception ignored in: <function _AutoFinalizedObjectBase.__del__ at 0x000001B5DC51BAE8>
Traceback (most recent call last):
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 84, in __del__
    self.finalize()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 144, in finalize
    self._finalizer()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\weakref.py", line 548, in __call__
    return info.func(*info.args, **(info.kwargs or {}))
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 104, in _do_finalize_object_ref
    obj._do_finalize_object()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\_objfinalizer.py", line 71, in _do_finalize_object
    self._finalize_object()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb1.py", line 604, in _finalize_object
    _lib.libusb_unref_device(self.devid)
OSError: exception: access violation writing 0x0000000000000024
```

Anyone got the same issue ? This exception raise when trying to parse url in ```open_mpsse_from_url() ``` function.

eblot, 2019-02-22:
I'm not sure it is related to pyftdi: it seems to be a pyusb issue. You may want to log this issue to pyusb project.

eblot, 2019-02-22:
If you do not call `terminate()`, does it help?

Stimaleger, 2019-02-22:
No it doesn't help. 
I can get it works if I configure the i2c controller using the open_mpsse method with providing vendor and product id. 

Gonna look into pyusb project. Thanks for help :blush:

eblot, 2019-02-22:
> I can get it works if I configure the i2c controller using the open_mpsse method with providing vendor and product id.

This is very weird. Why version of `pyftdi` are you using?

Stimaleger, 2019-02-22:
```bash
C:\Users\a.leger>pip3 show pyftdi
Name: pyftdi
Version: 0.29.3
Summary: FTDI device driver (pure Python)
Home-page: http://github.com/eblot/pyftdi
Author: Emmanuel Blot
Author-email: emmanuel.blot@free.fr
License: LGPL v2
Location: c:\users\a.leger\appdata\local\programs\python\python37\lib\site-packages
Requires: pyusb, pyserial
Required-by:
```

eblot, 2019-02-25:
What is the syntax you use with `open_mpsse` that actually works?

Stimaleger, 2019-02-25:
I add this method to the I2cController class : (gonna make a PR)

``` python
def configure_from_usb_id(self, vendor, product, **kwargs):
        """Configure the FTDI interface as a I2c master using USB ID

           :param int vendor: USB vendor id
           :param int product: USB product id
           :param kwargs: options to configure the I2C bus

           Accepted options:

           * ``frequency`` float value the I2C bus frequency in Hz
           * ``clockstretching`` boolean value to enable clockstreching.
             xD5 (GPIOL1) pin should be connected to xD0 (SCK)
        """
        for k in ('direction', 'initial'):
            if k in kwargs:
                del kwargs[k]
        if 'frequency' in kwargs:
            frequency = kwargs['frequency']
            del kwargs['frequency']
        else:
            frequency = self.DEFAULT_BUS_FREQUENCY
        # Fix frequency for 3-phase clock
        if frequency <= 100E3:
            timings = self.I2C_100K
        elif frequency <= 400E3:
            timings = self.I2C_100K
        else:
            timings = self.I2C_100K
        if 'clockstretching' in kwargs:
            clkstrch = bool(kwargs['clockstretching'])
            del kwargs['clockstretching']
        else:
            clkstrch = False
        ck_hd_sta = self._compute_delay_cycles(timings.t_hd_sta)
        ck_su_sta = self._compute_delay_cycles(timings.t_su_sta)
        ck_su_sto = self._compute_delay_cycles(timings.t_su_sto)
        ck_buf = self._compute_delay_cycles(timings.t_buf)
        ck_idle = max(ck_su_sta, ck_buf)
        self._ck_delay = ck_buf
        self._start = (self._data_lo * ck_hd_sta +
                       self._clk_lo_data_lo * ck_hd_sta)
        self._stop = (self._clk_lo_data_lo * ck_hd_sta +
                      self._data_lo*ck_su_sto +
                      self._idle*ck_idle)
        frequency = (3.0*frequency)/2.0
        self._frequency = self._ftdi.open_mpsse(
            vendor=vendor, product=product,
            direction=self._direction, initial=self.IDLE,
            frequency=frequency, **kwargs)
        self._tx_size, self._rx_size = self._ftdi.fifo_sizes
        self._ftdi.enable_adaptive_clock(clkstrch)
        self._ftdi.enable_3phase_clock(True)
        try:
            self._ftdi.enable_drivezero_mode(self.SCL_BIT |
                                             self.SDA_O_BIT |
                                             self.SDA_I_BIT)
        except FtdiFeatureError:
            self._tristate = (Ftdi.SET_BITS_LOW, self.LOW, self.SCL_BIT)
```

eblot, 2019-02-27:
Ok so that means the issue could be related to USB bus enumeration (UsbTools).

Which version of libusb is installed?

Stimaleger, 2019-02-27:
Version: 1.0.22b4

eblot, 2019-02-27:
Can you reproduce the crash with

    >>> from pyftdi.usbtools import UsbTools
    >>> UsbTools.find_all([(0x403, 0x6014)])

Stimaleger, 2019-02-27:
This function is not implemented for windows 10 :

``` console
Python 3.7.1 (v3.7.1:260ec2c36a, Oct 20 2018, 14:57:15) [MSC v.1915 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyftdi.usbtools import UsbTools
>>> UsbTools.find_all([(0x403, 0x6014)])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\pyftdi-0.29.3-py3.7.egg\pyftdi\usbtools.py", line 63, in find_all
    description = UsbTools.get_string(dev, dev.iProduct)
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\pyftdi-0.29.3-py3.7.egg\pyftdi\usbtools.py", line 498, in get_string
    return usb.util.get_string(device, strname)
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\util.py", line 311, in get_string
    langids = dev.langids
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 817, in langids
    self._langids = util.get_langids(self)
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\util.py", line 268, in get_langids
    0
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\control.py", line 173, in get_descriptor
    data_or_wLength = desc_size)
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 1025, in ctrl_transfer
    self._ctx.managed_open()
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\core.py", line 120, in managed_open
    self.handle = self.backend.open_device(self.dev)
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb1.py", line 786, in open_device
    return _DeviceHandle(dev)
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb1.py", line 643, in __init__
    _check(_lib.libusb_open(self.devid, byref(self.handle)))
  File "C:\Users\a.leger\AppData\Local\Programs\Python\Python37\lib\site-packages\usb\backend\libusb1.py", line 593, in _check
    raise NotImplementedError(_strerror(ret))
NotImplementedError: Operation not supported or unimplemented on this platform
```

eblot, 2019-02-27:
OMG. That's another reason I never wanted to support PyFtdi on Windows. I deeply hate this environment. I'll try to have a look but it is nightmare to work on.

Stimaleger, 2019-02-27:
Ahahah ^^ 
No problem dude thank you very much for your support but I'll keep going using open_mpsse solution. I'm usually working with Linux but I want to developp my application with cross-platform capabilities.
You want me to close the issue ?

eblot, 2019-02-27:
No I need to understand what's going on - and at least to reproduce it. 
Last time I tried to install pyftdi on Windows I lost one full day, I needed a lot of caffeine not to throw my computer away...
I probably need to add an API to bypass the USB bus enumeration anyway.

Stimaleger, 2019-02-27:
My setup : (if needed)
 - FT232H Adafruit <=I2C=>  ISL29125 RGB sensor

eblot, 2019-02-27:
Can you sum up how you installed libusb on Win10, I know that there are several ways (zadig, ..). to do it. Thanks.

Stimaleger, 2019-02-27:
Follow this instructions :
[Setup FT232h Adafruit on windows](https://learn.adafruit.com/adafruit-ft232h-breakout/windows-setup#libusb-driver-install-5-2)

Stimaleger, 2019-03-25:
Ok so it seems that this issue is due to libusb dll version ([see this issue](https://github.com/pyusb/pyusb/issues/203#issue-332146435)) 

There is no memory violation if you use [libusb v1.0.21](https://github.com/libusb/libusb/releases/tag/v1.0.21).

eblot, 2019-03-28:
Ok, thanks for the feedback. 
Could you give a try with libusb v1.0.22-final (released on March 25th)?

saraazizi76, 2019-08-25:
> Ok, thanks for the feedback.
> Could you give a try with libusb v1.0.22-final (released on March 25th)?

Hi i have some problems about installation of pyftdi.
i run these commands:

$sudo apt update
$ sudo apt install python3.6
$ sudo apt install python3-pip
$ sudo apt install python-pip
$ sudo apt install git
$ sudo pip3 install --upgrade setuptools
$ cd ~
$ sudo git clone https://github.com/eblot/pyftdi
$ cd pyftdi
$ sudo -H python3 setup.py install

but i get this error:
Traceback (most recent call last):
  File "setup.py", line 34, in <module>
    from setuptools import find_packages, setup
ModuleNotFoundError: No module named 'setuptools'

Do you know what is the problem?

eblot, 2019-08-25:
> Hi i have some problems about installation of pyftdi.

Sure, but it does not seem to be related with this ticket, please open a new issue.

saraazizi76, 2019-08-25:
> > Hi i have some problems about installation of pyftdi.
> 
> Sure, but it does not seem to be related with this ticket, please open a new issue.

right

sactosteve, 2020-11-06:
> Ok so that means the issue could be related to USB bus enumeration (UsbTools).
> 
> Which version of libusb is installed?

I'm seeing this problem as well.  I have a Miniconda3 installation; how do I get the libusb version?