eblot/pyftdi #122: bus error message trying to use i2c

btigner, 2019-02-24:
hardware: C232HM 5V ftdi cable FT232H chip on board
pyftdi 029.3 version
Python3.6 64bit
Running these commands in Python , I get a "bus error" and python crashes.
-------
from pyftdi.i2c import I2cController
i2c=I2cController()
i2c.configure('ftdi:///1')
(bus error)
--------
I am running on FreeBSD 13, I have set hw permissions to allow users to access devices, permissions
are correct, and lsusb (as user ) shows the cable is connected.

Bus /dev/usb Device /dev/ugen9.2: ID 0403:6014 Future Technology Devices International, Ltd FT232H Single HS USB-UART/FIFO IC

I get the error whether the cable is plugged in or not.

I do not have uftdi driver loaded so it should be using the ucom driver.

Running the commands as root produces the same message.

Any suggestions are welcome.

eblot, 2019-02-25:
It looks like an issue at usb level. I never tried using `pyusb` on FreeBSD.

Try to enable libusb debug (`export LIBUSB_DEBUG`...) to check if you can get more information.

btigner, 2019-02-26:
I did not get much debug info back.
What else can I do ?

in the sh shell i typed

$ export LIBUSB_DEBUG=3
$ python3.6
Python 3.6.8 (default, Jan 11 2019, 13:26:43)
[GCC 4.2.1 Compatible FreeBSD Clang 7.0.1 (branches/release_70 348686)] on freebsd13
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyftdi.i2c import I2cController
>>> i2c = I2cController()
>>> i2c.configure('ftdi:///1')
LIBUSB_FUNCTION: libusb_init complete
Bus error

$

eblot, 2019-02-26:
I'm afraid I do not have spare time to install FreeBSD for now and give it a try.
What version of `libusb` is installed (and loaded)?

It would be interesting to check if `pyusb` is able to communicate with `libusb` using another project, even a demo one.

A common issue I think about is a mismatch between what `pyusb` "thinks"/defines about the binary interface of `libusb`, and what is actually exported on FreeBSD. For example a pointer size mismatch could produce this early crash. This is why trying `pyusb` w/o `pyftdi` could give some clues.

HTH.

btigner, 2019-02-26:
It seems to work with pyusb.

->pip list |grep usb
libusb1                  1.7       
pyusb                    1.0.2 

->python3.6
import usb.core
dev=usb.core.find(idVendor=0x0403,idProduct=0x6014)
dev.set_configuration()
print(dev)

DEVICE ID 0403:6014 on Bus 009 Address 002 =================
 bLength                :   0x12 (18 bytes)
 bDescriptorType        :    0x1 Device
 bcdUSB                 :  0x200 USB 2.0
 bDeviceClass           :    0x0 Specified at interface
 bDeviceSubClass        :    0x0
 bDeviceProtocol        :    0x0
 bMaxPacketSize0        :   0x40 (64 bytes)
 idVendor               : 0x0403
 idProduct              : 0x6014
 bcdDevice              :  0x900 Device 9.0
 iManufacturer          :    0x1 FTDI
 iProduct               :    0x2 C232HM-EDHSL-0
 iSerialNumber          :    0x3 FT0NFAT4
 bNumConfigurations     :    0x1

+more

eblot, 2019-02-27:
Ok, I definitely need to have a look at this issue then.

eblot, 2019-02-27:
## macOS 10.14.3, pyusb 1.0.2, libusb 1.0.22

## FTDI device plugged in

    >>> from pyftdi.i2c import I2cController
    >>> i2c=I2cController()
    >>> i2c.configure('ftdi:///1')
    >>> from sys import platform
    >>> platform
    'darwin'

## Linux Mint 19.1 (Tessa), pyusb 1.0.2, libusb 1.0.21-2

### no FTDI device plugged in

    >>> from pyftdi.i2c import I2cController
    >>> i2c=I2cController()
    >>> i2c.configure('ftdi:///1')
    Traceback (most recent call last):
      File "<input>", line 1, in <module>
        i2c.configure('ftdi:///1')
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/i2c.py", line 363, in configure
        frequency=frequency, **kwargs)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/ftdi.py", line 471, in open_mpsse_from_url
        vendor, product, index, serial, interface = self.get_identifiers(url)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/ftdi.py", line 309, in get_identifiers
        cls.SCHEME, cls.VENDOR_IDS, cls.PRODUCT_IDS, cls.DEFAULT_VENDOR)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/usbtools.py", line 376, in parse_url
        urlstr)
    pyftdi.usbtools.UsbToolsError: No USB device matches URL ftdi:///1

### udev not configured

    >>> from pyftdi.i2c import I2cController
    >>> i2c=I2cController()
    >>> i2c.configure('ftdi:///1')
    Traceback (most recent call last):
      File "<input>", line 1, in <module>
        i2c.configure('ftdi:///1')
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/i2c.py", line 363, in configure
        frequency=frequency, **kwargs)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/ftdi.py", line 471, in open_mpsse_from_url
        vendor, product, index, serial, interface = self.get_identifiers(url)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/ftdi.py", line 309, in get_identifiers
        cls.SCHEME, cls.VENDOR_IDS, cls.PRODUCT_IDS, cls.DEFAULT_VENDOR)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/usbtools.py", line 342, in parse_url
        devices = devclass.find_all(vps)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/ftdi.py", line 390, in find_all
        return UsbTools.find_all(vps, nocache)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/usbtools.py", line 62, in find_all
        sernum = UsbTools.get_string(dev, dev.iSerialNumber)
      File "/home/eblot/Sources/Git/pyftdi/pyftdi/usbtools.py", line 498, in get_string
        return usb.util.get_string(device, strname)
      File "/home/eblot/.local/lib/python3.6/site-packages/usb/util.py", line 314, in get_string
        raise ValueError("The device has no langid")
      ValueError: The device has no langid

### udev configured

    >>> from pyftdi.i2c import I2cController
    >>> i2c=I2cController()
    >>> i2c.configure('ftdi:///1')
    >>> from sys import platform
    >>> platform
    'linux'

eblot, 2019-02-27:
BTW, I wonder what is libusb 1.7, as official libusb 1 versions range is [1.0.0 .. 1.0.23]?

eblot, 2019-02-27:
I can't get the network working w/ Parallels VM & FreeBSD 13, and the FreeBSD 13 QCOW2 image gets corrupted after a couple of reboot w/ QEMU, so it's going to be difficult to give it a try.

btigner, 2019-03-04:
I've been out for knee surgery, but I have some interesting results.
I've been trying different versions of pyftdi .
I can get it to work when I back off to pyftdi version  0.25.2 .
The error shows up when I use pip to uninstall pyftdi 0.25.2 
and install any newer version from 0.26.3 on.
my pip shows available versions of pyftdi as 
(from versions: 0.4.5, 0.5.1, 0.5.2, 0.6.3, 0.13.2, 0.13.4, 0.22.1, 0.25.0, 0.25.2, 0.26.3, 0.26.4, 0.27.0, 0.28.0, 0.28.2, 0.28.3, 0.28.8, 0.28.9, 0.28.10, 0.29.0, 0.29.1, 0.29.2, 0.29.3)
I haven't downloaded and diff'ed the code from pyftdi 0.25.2 and 0.26.3 yet.
Any suggestions would be appreciated.

btigner, 2019-03-04:
the lisbusb1.1.7 was installed from a system package
py36-libusb1-1.7               Pure-python wrapper for libusb-1.0
I uninstalled it and got the same result.
pip search libusb shows 
libusb (1.0.22b4) 
I installed it using pip and still got the
same results, it works with pyftdi <= 0.25.2
and does not work with pyftdi > 0.25.2 .
So, I can at least get the code working with pyftdi 025.2.

eblot, 2019-03-28:
Could you give a try with libusb 1.0.21 or 1.0.22-final? It seems there is an issue with the libusb beta version, see https://github.com/pyusb/pyusb/issues/203#issue-332146435

nelsoncsc, 2019-10-04:
Hi,
I have a similar problem with FT232H in Ubuntu 18.04. When I run  **i2c.configure('ftdi:///1')** I got the "device has no langid" error. I installed libusb, installed the D2xx and created udev rules from https://www.element14.com/community/docs/DOC-76656/l/udev-rules-file-for-ftdi-ft2232dh-ft232h-and-papilio-duo.

eblot, 2019-10-05:
@nelsoncsc I see no relationship with your issues and this (closed) ticket.

I do not understand why you would have to install d2xx.
Have you read the installation [documentation](https://eblot.github.io/pyftdi/installation.html)? 

In any case, open a new ticket when the issue is different, thanks.

nelsoncsc, 2019-10-05:
Hi @eblot ,
sorry, I was a bit confused first, there is no need to install d2xx. I created a new issue for a different problem.
Thank you

txmx, 2022-05-16:
experiencing this issue on FreeBSD 12.2, when trying to interact with Tigard board (FT2232H)
**python** 3.8.12
**pyusb** 1.2.1
**pyftdi** 0.54.0
**libusb** distribution is shipped with the OS, version 1.0

`sudo ./i2cscan.py` emits `Bus error` without any further information when device is plugged in; emits `No USB-Serial devices detected` when device is absent.