eblot/pyftdi #125: SPI clock frequency selection is too limited

vestom, 2019-04-30:
I tried using SPI with a frequency of 2.5 MHz with a FT232H for a timing critical applications (PWM generation), but could not get it to work. It turned out, that the frequency was set to 2 MHz instead, although the hardware is capable of 2.5 MHz (30 MHz / 12).

When inspecting the code for Ftdi._set_frequency(), a couple of issues were apparent:
- When the desired frequency is below BUS_CLOCK_BASE (6 MHz) it will aways run with pre-divider enabled, limiting available frequencies to ratios of 6 MHz (3, 2, 1.5, ...), even when hardware is capable of a more accurate frequency ratios of 30 MHz (..., 6, 5, 4.3, 3.75, 3.33, 3, ...).
- Divisor is always rounding up, even when closer higher frequency is available (e.g. 5.9 MHz => 3 MHz). It is of course a choice, however in many cases it would more likely to be within acceptable tolerances if rounding towards frequency with lowest percentual error.
- For extremely low frequencies (below ~92 Hz), the divisor will overflow 0xFFFF and wrap around.

A possible fix is suggested in #124.

eblot, 2019-04-30:
Merged in e2c96778e5. Thanks!

closed 2019-04-30 by eblot

vestom, 2019-04-30:
Thanks to you also. It has been a pleasure making a small difference for this well organized code base :-)