eblot/pyftdi #126: Decreasing time between SPI writes?

nikhil9dua, 2019-05-14:
Hi there,

My question have been asked before related to the I2C delay, however, I'm writing to a SPI based DAC which takes in a single write sequence as 3 bytes and updates the DAC output on CS rising edge. 

I can write these bytes using command below (test_dac.slave is a instance of SpiController().get_port):
CH A:
 test_dac.slave.write([0x08, 0x19, 0x99]).tobytes() 
CH B:
 test_dac.slave.write([0x09, 0x19, 0x99]).tobytes()

I measured the delay between these two writes and is its roughly about 200 uS, which is extremely high for my application, even when I run the SPI clock at 25 MHz.  

Because I have to pull the CS bar high every 3 bytes, is there a faster way of doing this, where the CS bar remains high for less time instead of 200us??

[image]

eblot, 2019-05-15:
I do not think PyFtdi is the way to go, and if you want to achieve a precise time control, even FTDI is not.

You could customize PyFtdi to prepare several SPI requests and send them into a unique buffer, so that the FTDI device could streamline the SPI master request on the SPI bus. But:

When the FTDI buffer is full (which depends on the actual FTDI device, I guess you could fit from 100 up to 400 3-byte transfer in a single buffer), you will hit a hard-to-control delay. Moreover there is no "NOP" instruction on FTDI, so it would also be quite difficult to execute the SPI commands with a known/configurable pace. Said otherwise, I'm not sure you can ever achieve some kind of real-time/time-controlled streaming over USB & FTDI.

If you really need to send commands over the USB bus to a SPI controlled device at a controlled pace, I would recommend you pick another SPI master than FTDI's.

For about the same price, I would opt in for a STM32 microcontroller for example. A Nucleo32 board is about $10, and is far more flexible - at the expense of custom ARM code.  I think Cypress also provide some USB bridges with (ugly 8051-based) embedded microcontrollers :-)

nikhil9dua, 2019-05-15:
Ah gotcha. 
Changing a host will be tough since my whole system is designed around the FT232H :) but I will most definitely take your consideration for next boards. Will an arduino be sufficient, to reduce complexity of writing custom ARM code you think? 

In the mean time, I can try your suggested method. How would I send multiple SPI requests using PyFtdi using the FTDI buffer? I think even 200 3-byte transfers should be sufficient for my application, as my DAC only changes output when I write a 3-byte value and holds the previously written value, so I'd only need to send data when transitioning.

Thanks again!

eblot, 2019-05-16:
I guess an Arduino could be enough yes (it all depends on the actual bitrate you need to achieve) - I do not like Arduino at all, but I do not see why it would not work.

Back to PyFTDI: around spi.pi@570, instead of calling `_write_data`, you could push the prolog/cmd/epilog into a bytearray-like buffer, and add write the whole buffer at once - when it reaches some predefined depth or with an explicit API to do so. I'm not sure how this would behave when you reach the FTDI HW FIFO size - I guess it would work but you may expect variables delays between some SPI writes.

Let me know how it goes.

nikhil9dua, 2019-05-20:
I've tried and was able to get this partially working! 

However, I've noticed that the write data, say for example, is [0x08, 0x19, 0x99] the byte array object is 21 characters long --> array('B', [128, 253, 251, 128, 245, 251, 16, 2, 0, 8, 25, 153, 128, 245, 251, 128, 253, 251, 128, 248, 251]) where the numbers 8, 25, 153 are my actual write command 

Are the rest of the numbers in this byte array the usb core write information? I can write two back to back commands here if I just append two of these array as per below. Then in my example [0x08, 0x19, 0x99, 0x09, 0x19, 0x99] would just be appended as two copies of the same 21 letter byte array with just the 8 switched with 9 in the middle as per below, with the sequence starting as 128, 253, 251:

array('B', [128, 253, 251, 128, 245, 251, 16, 2, 0, 8, 25, 153, 128, 245, 251, 128, 253, 251, 128, 248, 251, 128, 253, 251, 128, 245, 251, 16, 2, 0, 9, 25, 153, 128, 245, 251, 128, 253, 251, 128, 248, 251])
 
My only issue now is issuing a wait statement. Although I can get two back to back of these commands to work, what would be the best way to have a controlled wait statement inserted between these two commands, say  a 100uS.  The way that I have tried is just keeping on appending the same byte array with garbage 3 byte writes to DAC so it doesn't change but I noticed that noise becomes extremely high due to this and and I add more than say 10 of these garbage 3-byte writes,  I start getting USB errors as you mentioned. Any other alternatives?? My DAC will just spit out the last value written in it's register so I just need to "hold that value" in a timed fashion. 

Thanks so much again!

eblot, 2019-05-21:
The bytes you see are the MPSSE FTDI dialect, ''i.e.'' all commands that are required to assert /CS, manage clock, clock data out, etc. These bytes are encapsulated into USB, you cannot see those bytes in the trace.

As I said earlier, there is no "NOP" operation on FTDI, which makes adding delay between packets quite impossible, as each available FTDI command results in action on FTDI I/O. Your best chance, if your HW supports it - that is stays idle if clock signal beats while /CS is not asserted, is to add clock-without-data commands between real commands.

Again, I'm not convinced you can achieve anything really robust and reliable with an FTDI if you need to satisfy real-time constraints...

nikhil9dua, 2019-05-21:
I will try that. If not, I'll just stick to my original timing constraints. Thanks for your help again!!

JorgezGlue, 2019-12-12:
> When the FTDI buffer is full (which depends on the actual FTDI device, I guess you could fit from 100 up to 400 3-byte transfer in a single buffer), you will hit a hard-to-control delay. Moreover [...]

I'm doing a similar exercise with the MPSSE JTAG and I'm also running into the delay when the FTDI buffer is filling up. Do you know more details pertaining this delay? Is it because of the inherent python overhead or is it a hardware limitation?

I read the following from the FT232H datasheet as an example:
"USB to synchronous 245 parallel FIFO mode for
transfers up to 40 Mbytes/Sec"
But I don't think this applies to MPSSE.

Any insights that you can provide would be greatly appreciated.

Thanks!

eblot, 2020-01-29:
FTDI does not provide any documentation about how their chips behave, so it is hard to tell.
40 MB/s is the peak transfer rate on a USB2 HS bus. It does not mean the transfer occurs at a constant pace though - and it is likely not achievable with a real use case anyway...