eblot/pyftdi #127: the clock waveform output problem in SPI mode 3.

nxdhf, 2019-05-16:
Hi, My device is FT232H, I use it as an SPI master to communicate with other devices.
The clock output is incorrect when I set it to mode 3.
Here is the screenshot. The signal is MOSI, MISO, CS, SCK from top to bottom.

[image]

Here is my code:
```python
spi = SpiController(cs_count=1)
spi.configure('ftdi:///1')
slave = spi.get_port(cs=0, freq=100*1000, mode=3)
slave.write([0xA6])
slave.write([0x6A])
jedec_id = slave.read(20)
print(jedec_id)
```

eblot, 2019-05-16:
I think you hit a limitation of the FTDI HW itself.

From FTDI Application Note 114, v1.1 page 3, you can read:

"_It is recommended that designers review the SPI Slave data sheet to determine the SPI mode implementation. FTDI device **can only support mode 0 and mode 2** due to the limitation of MPSSE engine._"

The PyFTDI code says that is mode 2 that is not supported, but it is likely wrong. The fact is that I never add mode 2 & mode 3 -only SPI slaves to check what actually worked, and when I wrote this part of the code, I did not have a logic analyser available to verify the actual output. Now that I own one, I lack time to work on those modes.

nxdhf, 2019-05-16:
Thank you for your response. This library has helped me a lot.
On this issue, for example, [FTDI Application Note 108](https://www.ftdichip.com/Support/Documents/AppNotes/AN_108_Command_Processor_for_MPSSE_and_MCU_Host_Bus_Emulation_Modes.pdf) Chapter 3.3.4, It mentions the command output on the falling edge of the clock. This is also where I am confused.
As I understand it, this library sends commands to The MPSSE Command Processo by driver, right?
Maybe I have to dive into the code and see if the command is correct.

eblot, 2019-05-16:
> his library sends commands to The MPSSE Command Processo by driver, right?

Yes. Check out `spi.py`

eblot, 2019-10-28:
Have a look at the master branch, I've started to fix some long standing SPI issues, and it may help with mode 3. I do not have any mode 3 slave, so it is hard to tell with its works as expected, but the logic analyser capture I made seem ok, at least in the MOSI direction.

eblot, 2020-02-07:
Feel free to re-open if current version does not work with mode 3 (keep in mind FTDI does not officially support mode 3)

closed 2020-02-07 by eblot