eblot/pyftdi #128: Wrong frequency value returned by I2cController.frequency

marccheneau, 2019-06-03:
Hi there.

First of all I wish to express my gratitude for the amazing job! I am a newbie to programming embedded devices and I intend to stay so as it's not my core activity. Your incredibly easy-to-use library just spared me weeks of works!

To the fact now: when configuring the frequency of an I2C controller, the value specified by the user is multiplied by 3/2 internally before being sent over to Ftdi.open_mpsse_from_url. I guess there is a good reason for that.
But when I want to read out the frequency after configuration, I would have expected to get back the "physical" frequency. For instance, if I configure the controller for using 100 kHz and then read out the configured value, I get 150 kHz instead of 100 kHz.
Or am I missing something?

eblot, 2019-06-03:
Hi, thanks -  a lot.

Well, I think that's a bug - the frequency you read back out.

The reason for the 3:2 ratio is because of the way the FTDI works. A 3-phase clock is used for I2C bus feature, and the FTDI clock needs to be modified so that the SW requested frequency is actually achieved on the physical I2C bus.

I may have forgotten to update the read back API to manage this kludge, I will fix it. Thanks for the report.

marccheneau, 2019-06-03:
Ok, great. Thanks for the explanation and for the quick reply!

eblot, 2019-06-03:
If you could give a try to the last version, as I do not have the HW at hand to test it... Thanks.

closed 2019-06-03 by eblot

marccheneau, 2019-06-03:
Sure. Right away.

marccheneau, 2019-06-03:
I just tested the last version and it seems to do the job. I could talk to an I2C slave device and I do get back the frequency I configured. Thanks.

eblot, 2019-06-03:
Ok, thanks for the feedback, and good catch!

marccheneau, 2019-06-03:
You're welcome! A tiny contribution to a big and great project ;)

commit 0163bf7b (2019-06-03): Fix #128: wrong I2C frequency readback