eblot/pyftdi #129: CS Pinout Usage in SPI Mode

dua564, 2019-06-07:
Hi there,

I have the Adafruit FT232H using pyftdi library that is connected to 5 SPI slave devices. The CS lines are connected to various pins such as ACBUS4, ACBUS5, ACBUS6 etc..

 However, I noticed that the pyftdi pinout only allows CS lines on ADBUS3-ADBUS7  (as CS 0 to 4) in order. Is this true?  

I'm able to specify cs_count=5 when calling SpiController. 

Example, my ADC is connected on ACBUS6 but when I use get_port(cs=10) for CS pin on ACBUS6, I get an error stating "no SPI port".   How would I be able to call this ? 

Thanks!

Nik

eblot, 2019-06-07:
Yes this is a limitation that has not been addressed yet: /CS lines should be on the same physical port as the SPI bus (that is CS lines/slaves max=4 for a 232H). Up to now, there has been no real requirement to support more that 4 slaves.

It should not be a major change to the code. The truth is that there are so many combinations and so many ways to use SPI (and at a smaller range I2C) buses that it has started to be difficult to validate every possible combination. Which means that I'm reluctant to add more features without a way to test it and more importantly to check for no regression with the existant code base. Every single slave behaves slightly differently than the other.

I've therefore started to work on a I2C+SPI slave configurable device which could help testing those combinations, but this project is in a early state. Once it is completed, I will have far more confidence to add features so the actual PyFtdi implementation can be really - and hopefully deeply - validated.

Back to this limitation - that I should have documented - the MPSEE command to drive upper 8 bit is different from the lower 8 bits, so the code should detect which byte the selected /CS should address, and generate the MPSSE sequence accordingly. It is already done for the SPI-GPIO part of the code, so that should be quite easy to add, if you want to have a look at the code.

dua564, 2019-06-11:
Hi eblot,

Thanks for your response! 

Is the supported slaves 4 or 5? Currently on my board, I have 5 slaves hooked up to just a standalone SPI Controller Adafruit FT232H

As for the MPSEE command, would I just have to modify the CS_BIT = 0x08 to something else in SpiController? I wasn't able to find so in the SPI-GPIO part of the code. Perhaps you can guide me as I'm still a bit of a beginner with dealing with this type of low-level logic :) 

Thanks!

-Nik

eblot, 2019-06-11:
Sorry, 5 slaves should work (with cs_count=5), there are 8 bits available on the first port so:
SCLK/MISO/MOSI, 5 pins are left for /CS

If you need more than 5 slaves, it's gonna be more complex, as the command to drive the /CS lines will differ whether /CS is <= or > 5.

eblot, 2019-08-29:
SPI CS lines up to 5 should be supported with the current release.

closed 2019-08-29 by eblot